openscad/openscad #2744: localization causing massive slowdowns on Windows

thehans, 2019-01-30:
This is more followup from the OpenSCAD forum thread by @nophead, trying to diagnose the slowness of preview for this script: https://gist.github.com/nophead/fd4ca58b5f6b19dc25bb98f7edbae90f

I tried profiling the build from CircleCI in Windows, and found that most of the time is being spent in `msvcrt.dll!setlocale`  
[image]
The build doesn't have debug symbols so I can't see much about the execution path within OpenSCAD code that is leading to this, but un-checking Preferences -> Advanced -> "Enable user interface localization" makes it comparably fast to the linux builds I have tested.

thehans, 2019-01-31:
I could really use some assistance getting a windows build with debug symbols enabled.  Can we get CircleCI to spit one out, maybe on a separate "windows_dev" branch or something?

nophead, 2019-01-31:
I have returned to a large project I last played with March 18. Compiling
seems to have got massively slower. When it has been previewed once I
expect F5 to take about 20 seconds to run through everything and decide it
hasn't changed. It seems to be taking more like 4 minutes now.

nophead, 2019-01-31:
Now down to 11 seconds, my fault, not OpenScad.

thehans, 2019-02-01:
I got a debug build on Windows and I can see that `setlocale` is being called by `std::__convert_from_v(int* const&, char*, int, char const*, ...) ()`, which apparently happens whenever a floating point number is output to an ostream.  

I also found this short discussion which seems relevant, although I have to admit I don't fully understand:
https://stackoverflow.com/questions/22753707/is-ostream-operator-in-libstdc-thread-hostile

Here are the first few lines of stacktrace which seem to be the most called: 
```
#0  0x00007ffb99c59c19 in setlocale () from C:\Windows\System32\msvcrt.dll
#1  0x0000000001d8b801 in std::__convert_from_v(int* const&, char*, int, char const*, ...) ()
#2  0x0000000001c9c8f4 in std::ostreambuf_iterator<char, std::char_traits<char> > std::num_put<char, std::ostreambuf_iterator<char, std::char_traits<char> > >::_M_insert_float<double>(std::ostreambuf_iterator<char, std::char_traits<char> >, std::ios_base&, char, char, double) const ()
#3  0x0000000001c9d53f in std::num_put<char, std::ostreambuf_iterator<char, std::char_traits<char> > >::do_put(std::ostreambuf_iterator<char, std::char_traits<char> >, std::ios_base&, char, double) const ()
#4  0x0000000001cb428d in std::ostream& std::ostream::_M_insert<double>(double) ()
#5  0x0000000001baa6f4 in std::ostream::operator<< (__f=<optimized out>, this=0x174e76d0) at /mxe/usr/lib/gcc/x86_64-w64-mingw32.static.posix/7.3.0/include/c++/ostream:221
#6  tostring_visitor::operator()[abi:cxx11](double const&) const (this=this@entry=0x174e78bf, op1=@0x2fd42b28: 3548) at ../src/value.cc:246
```
full trace attached
[windows_stacktrace.txt](https://github.com/openscad/openscad/files/2819859/windows_stacktrace.txt)

I should also note that even with localization turned off, I can still see a second or so of CPU time(from an 11s preview) going to `setlocale`, which is being called from:
 `__adjacent_find<__gnu_cxx::__normal_iterator<long long unsigned int*, std::vector<long long unsigned int> >, __gnu_cxx::__ops::_Iter_equal_to_iter>`

[image]

I don't really know what to do with this information at this point.  Anyone more familiar with how localization works can chime in?

thehans, 2019-02-01:
I guess the thing that struck me was the response on the stackoverflow saying that its not an issue for glibc > 2.2 ?  
Is there a way to see what version the windows builds are using and if they are older can we update them?  Also confusing that the initial guy is talking about libstdc++, but the other guy is mentioning glibc which is C only I thought?

kintel, 2019-02-01:
Help->Library Info should show the GLib version.
Not sure where GLib would be involved in this

kintel, 2019-02-01:
There is some slight weirdness in the double serialization code. You could try "fixing" that function and see if that changes the performance profile: https://github.com/openscad/openscad/blob/master/src/value.cc#L242

kintel, 2019-02-01:
wops, right.

It looks like we've linking with msvcrt, not glibc. ..which is a bit odd since we build with gcc. I have not idea how the mingw toolchain is stitched together on Windows, but could it be that there are different build variants behaving in different ways? The official binaries are built using MXE, which I guess never touches msvcrt, but the msys2 build is built on Windows-native and may behave differently.

thehans, 2019-02-01:
Hmm, I just found out about this project: https://github.com/google/double-conversion
I'm tempted to try integrating that. 

It could be ~14x speedup over `ostringstream` currently being used, based on this benchmark https://github.com/miloyip/dtoa-benchmark

nophead, 2019-02-01:
This is the MXE version that we are disusing. This is what I run
nowadays:  ci/circleci:
openscad-mxe-64bit unless I build it myself with MSYS2. Even then I tend to
download the MXE version from my own PRs because the MSYS2 version doesn't
run outside of MSYS2 because it needs lots of DLLs.

thehans, 2019-02-01:
So anyways, @kintel what do you think about the possibility of using google's [double-conversion](https://github.com/google/double-conversion) library?  

Its basically from their V8 engine and optimized for efficiency, I think it could be a nice improvement for us, I'm just not sure about optimal way to go about including it.  Would it be best for OpenSCAD org to make a fork and then git submodule that?

thehans, 2019-02-01:
Oh, it is actually packaged in Ubuntu and Debian as `libdouble-conversion1` and `libdouble-conversion-dev`

t-paul, 2019-02-01:
But it's needed mainly for Windows? It does not seem to be in MXE or in MSYS2. With packages in Linux distros we might have the option to include it for Windows (and maybe even for MacOS) but ignore and use the distro version for Linux.

thehans, 2019-02-01:
Well I did find this: https://github.com/Alexpux/MINGW-packages/tree/master/mingw-w64-double-conversion

Not sure how useful that is, I'm still a little confused about how the overall mingw build process works.

I think all platforms could benefit from more efficient double to string operation though, particularly in a use case like this with many polyhedron vertices.

kintel, 2019-02-02:
It's worth trying. Is there an easy way to test this through out CI or otherwise.
For such a small library, we could always just vendor it, but that's not something Debian & al likes very much..

thehans, 2019-02-02:
Interestingly I also found that Qt5 includes it as part of core:  https://doc.qt.io/qt-5/qtcore-attribution-doubleconversion.html
Not sure if that is of any use to us though.  I'm guessing that we don't want `src/value.cc` depending on Qt?  Is that why it was initially moved away from QString as the code comment mentions? https://github.com/openscad/openscad/blob/master/src/value.cc#L242
```
    // attempt to emulate Qt's QString.sprintf("%g"); from old OpenSCAD.
```

kintel, 2019-02-02:
Yup, the first iteration of the backend used Qt datatypes all over the place.

nophead, 2019-02-02:
Would a quick hack be to use the QT version when linked with QT and not
when built standalone?

kintel, 2019-02-02:
That's a very quick hack. Good for testing though

t-paul, 2019-02-02:
Hacking in that double-conversion library sounds much more promising to me.

nophead, 2019-02-02:
What is the non QT build actually used for?

thehans, 2019-02-03:
Hey, so I've done some work on testing out this library, and I've got it to build on my Linux machine.  
I have the code up on this branch if you want to see it: https://github.com/thehans/openscad/tree/double-conversion
The branch is based off of the changes I already made in https://github.com/openscad/openscad/pull/2738 and and continued from there.
Right now I'm guessing it won't build on any of the CI servers because I haven't dug much into package configuration etc.  For now I just manually installed `libdouble-conversion1` and `libdouble-conversion-dev` prior to building.

Without any changes to the test scripts, the double-conversion branch passes everything except for `/testdata/scad/functions/rands.scad` where it looks like its only changing results slightly when ridiculous infinite ranges are passed to `rands` (the numbers look more random than the 'expected' ones actually, since those all contain a bunch of 999999 digits after about the sixth most significant figure).

However, there is code in `tests/test_cmdline_tools.py` which is post-processing the output, so that hides a lot of what the library is really doing.
If I comment out this section, there's a lot more failing tests:
https://github.com/openscad/openscad/blob/d8a6ef32a590904478c6a4cf6a7346e6d7cf7d1d/tests/test_cmdline_tool.py#L109-L116

I'm attaching the resulting logs I get.  Mostly the results are just a few more digits of precision coming from libdouble-conversion, but there are some annoying almost "round" numbers that I feel like it should handle better.
Here's a short snippet of some a test with some of those:
```
 ECHO: v = 0, op = "v[4]", result = undef
 ECHO: v = -4.2, op = "not v", result = false
 ECHO: v = -4.2, op = "-v", result = 4.2
-ECHO: v = -4.2, op = "v *", result = -12.6
+ECHO: v = -4.2, op = "v *", result = -12.600000000000001
 ECHO: v = -4.2, op = "* v", result = -8.4
-ECHO: v = -4.2, op = "v /", result = -1.4
-ECHO: v = -4.2, op = "/ v", result = -0.47619047619
-ECHO: v = -4.2, op = "v %", result = -1.2
+ECHO: v = -4.2, op = "v /", result = -1.4000000000000001
+ECHO: v = -4.2, op = "/ v", result = -0.47619047619047616
+ECHO: v = -4.2, op = "v %", result = -1.2000000000000002
 ECHO: v = -4.2, op = "% v", result = 2
-ECHO: v = -4.2, op = "v +", result = -1.2
+ECHO: v = -4.2, op = "v +", result = -1.2000000000000002
 ECHO: v = -4.2, op = "+ v", result = -2.2
 ECHO: v = -4.2, op = "v -", result = -7.2
 ECHO: v = -4.2, op = "- v", result = 6.2
```

[TestResults.zip](https://github.com/openscad/openscad/files/2825107/TestResults.zip)
There are 3 reports zipped up.  One shows the double-conversion branch WITH substitutions (meaning `tests/test_cmdline_tools.py` is still post-processing the test data).  Another with double-conversion branch with no substitutions.  
And lastly the master branch with no substitutions for comparison.  (master branch with substitutions passes all tests of course, no need to include)

I'm using `ToShortest` to convert the numbers, the documentation for which is all in the `double-conversion.h` header
https://github.com/google/double-conversion/blob/87d21e3fee5a8cd3928bf093fbf0787e41dd4830/double-conversion/double-conversion.h#L134-L159

There are other options like `ToFixed`, and `ToPrecision` but these return a fixed number of digits after the decimal, or fixed total number of significant digits respectively, regardless if its just a bunch of zeroes after the decimal.  There's also `ToExponential` which claims to output the "shortest" value if you request -1 digits, but then every number is of the form `X.YeZ` whigh isn't great either.

Not really sure if there's any way to get it to behave nicely without basically re-wrapping a bunch of its lower level functions with custom code, which seems rather daunting.

Oh and lastly I should mention that the performance boost is definitely real.  The optimized code I was working on for #2737 was already down to 9s displaying 400 springs before using this library, and now its down to 6s.  Again this is on linux which wasn't particularly experiencing slow-downs compared to windows.

nophead, 2019-02-03:
I don't think the "round" numbers are actually round in floating point. I.e. 1.2 certainly isn't. 

It looks like you are simply outputting more digits. I think outputting less would round the last digit and fix the results. 

Before your change echo certainly doesn't output the full set of digits from a double. E.g. echo(1/3) gives
ECHO: 0.333333

kintel, 2019-02-03:
@thehans Nice! One note: I believe the code in `test_cmdline_tools.py` is there to account for different rounding modes across OS or libc environments.

thehans, 2019-02-03:
> I don't think the "round" numbers are actually round in floating point. I.e. 1.2 certainly isn't.

Yeah I understand that floating point has limitations in terms of representing decimals.  Was hoping there was some way that the library could truncate for *very nearly* those sort of values, but I think its going more for exact representations such that double to string and back to double is equal to the start value (for `ToShortest` mode anyways).

> It looks like you are simply outputting more digits. I think outputting less would round the last digit and fix the results.

The problem was that `ToShortest` which I was trying to use doesn't have any option to specify how many digits.  You just get what you get, and no way to round its output string.  

> Before your change echo certainly doesn't output the full set of digits from a double. E.g. echo(1/3) gives
> ECHO: 0.333333

Yes it appears the default before changes is for 6 total digits of precision(leading zeroes don't count).  I was able to write a function to trim trailing zeroes after decimal point and switched to `ToPrecision` mode, and set it to 6 digits for consistency.  The trim function feels like a bit of a mess, but I was going for max efficiency of iterating through the output characters no more than once.  I just pushed those changes to https://github.com/thehans/openscad/tree/double-conversion

> @thehans Nice! One note: I believe the code in `test_cmdline_tools.py` is there to account for different rounding modes across OS or libc environments.

OK, the `ToPrecision` mode *does* round the last digit.  I think with this library it should be consistent across platforms, so we may have the option to remove that `test_cmdline_tools.py` post-processing code if we update the expected tests.  As I mentioned above I've set it to 6 digits precision for now, but we do have the option to increase precision a little bit if wanted.  I have noticed that `ToPrecision` is actually much slower than `ToShortest` for more than 8 digits.  Using the test sample of previewing 400 springs I get 6s for 6 digits, 7s for 7 digits, 8s for 8 digits, then it jumps to 14-15s for 9 digits and beyond; it seems to branch to a different algorithm for those higher precisions, which is actually slower than `ostringstream` on Linux anyways (baseline 9s for same benchmark before double-conversion implemented).   

If high precision is desirable at all, I have an idea that I might be able to modify my `trimTrailingZeroes` to work well with `ToShortest`, which seems to be significantly more efficient than `ToPrecision` for those higher digit counts.

thehans, 2019-02-03:
Also annoyingly the `ToPrecision` mode doesn't handle special values `inf`, `nan`, whereas `ToShortest` does, and the `HandleSpecialValues` function that `ToShortest` calls is private so I had to duplicate its functionality.

thehans, 2019-02-04:
The Travis and AppVeyor builds are all passing now.  Yay!  

Just need to get CircleCI working now.  As far as I understand there are custom docker images that need to be updated for that?  I'm not sure there's anything I can do to update those.  Is the `scripts/docker-build.sh` used to create that?  I see that it calls `uni-get-dependencies.sh` which should work on Ubuntu/Debian at the very least.  I don't see anything about MXE in there though, so is that just for the AppImage?
As far as I can tell, double-conversion is not a prebuilt package for MXE, I've only found this which I'm hoping is useful:
http://hg.octave.org/mxe-octave/file/c2ac58e063b5/src/double-conversion.mk

Also I wasn't really sure if I needed anything special in the `.prf`, but my minimalist effort seems to be working ok so far:
 https://github.com/openscad/openscad/blob/da0cb42a1b12f2a82a238e22edb682a529c0c7fc/features/double-conversion.prf

kintel, 2019-02-04:
The dockerfiles for the MXE images are here: https://github.com/openscad/docker-openscad

They are split up a bit since it's really time consuming to build the images, and we have to maintain two CPU variants, so it may look a bit messy..

thehans, 2019-02-04:
Thanks, I was able to get it working.  That `.mk` file was junk and had to be re-written, but besides that it was a pretty smooth process to add it once I realized that the docker images run from openscad's own MXE fork.

https://circleci.com/gh/openscad/openscad/1482#artifacts
Running test script with `n=20` (400 springs)
With localization preference on: 8s
With localization preference off: 7s

Compared to latest build on master: https://circleci.com/gh/openscad/openscad/1479#artifacts
With localization preference on: 77s
With localization preference off: 28s

t-paul, 2019-02-05:
Nice, I'll try to upstream our changes to MXE once things quiet down a bit. That should make the snapshot build branch just track MXE master again...

thehans, 2019-02-06:
Closing since PR #2754 has been merged.

closed 2019-02-06 by thehans