openscad/openscad #2746: Crash - lots of objects

MichaelAtOz, 2019-01-31:
[image]

Latest snapshot 2019.01.24.ci1256 (git 7fa2c8f1) & 2015.03-2, Win64.

Code, n=200 works, n=210 doesn't.
```
module lots(n=250) 
  for (i=[0:n-1],j=[0:i]) 
    translate([i,j,0]) 
      if (true)
        cube(1);
      else
        sphere(1);
 
lots();
```
originally was sphere, tried cube, so it seems object count related not facets??

thehans, 2019-01-31:
This is due to stack size limitations.  The changes in #2343 let it go up to n=500-600 or so.  Beyond that it will preview once, but if you preview again it will crash due to some kind of recursion in the `shared_ptr` destructors. 

Also these limits can be increased by changing the max stack size.  On Linux `ulimit -S -s 65536` for example.  On windows it looks like its determined by the OpenSCAD build, set to a default 8MB, to match with linux defaults.
https://github.com/devlaam/openscad/commit/7a04063c74af00d92bcaf3f7476e95cc645fd797

Here is what stacktrace looks like on #2343 branch if you go beyond those limits:
```
#0  0x000055555563e73f in std::_Sp_counted_ptr<CSGOperation*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() ()
#1  0x000055555563e742 in std::_Sp_counted_ptr<CSGOperation*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() ()
#2  0x000055555563e742 in std::_Sp_counted_ptr<CSGOperation*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() ()
#3  0x000055555563e742 in std::_Sp_counted_ptr<CSGOperation*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() ()
#4  0x000055555563e742 in std::_Sp_counted_ptr<CSGOperation*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() ()
#5  0x000055555563e742 in std::_Sp_counted_ptr<CSGOperation*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() ()
#6  0x000055555563e742 in std::_Sp_counted_ptr<CSGOperation*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() ()
#7  0x000055555563e742 in std::_Sp_counted_ptr<CSGOperation*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() ()
#8  0x000055555563e742 in std::_Sp_counted_ptr<CSGOperation*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() ()
#9  0x000055555563e742 in std::_Sp_counted_ptr<CSGOperation*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() ()
#10 0x000055555563e742 in std::_Sp_counted_ptr<CSGOperation*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() ()
#11 0x000055555563e742 in std::_Sp_counted_ptr<CSGOperation*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() ()
#12 0x000055555563e742 in std::_Sp_counted_ptr<CSGOperation*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() ()
#13 0x000055555563e742 in std::_Sp_counted_ptr<CSGOperation*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() ()
...continuing ad nauseum
```

MichaelAtOz, 2019-01-31:
Splitting into 2 `for()`'s goes higher

```
module lots(n=950) 
  for (i=[0:n-1]) 
    for (j=[0:i])
    translate([i,j,0]) 
      if (true)
        cube(1);
      else
        sphere(1);
 
lots();
```
I suspect the double-barrel* `for()` could be serialised to not use stack?   * or n-barrel

I got to n=2000 which is sufficient for my needs ATM. Preview & GUI was not very responsive :/

t-paul, 2020-04-27:
I believe this is fixed via #2343 - I can render a single loop with `n = 2000`.

closed 2020-04-27 by t-paul