openscad/openscad #2755: $linenumber and $currentfilename $mainfilename variables or functions & date & time

MichaelAtOz, 2019-02-04:
re #2633 

> If someone has a good use case for linenumber, currentfilename and/or mainfilename, please write and example and open an issue - I am happy to revisit this topic when I see a good use case.

Below shows $variables and now() [for date-time], but I don't care about the mechanism, just access to data in some systemic way, it could be a system() function or some such, e.g. `system("date","YY MMM DD")`, `system("currentfilename")`.

[image]

While this is one source file, consider it in context of libraries and with version numbers in filenames.
Here is a snippet of some of my files
[image]
Being able to stamp the STLs with the source would be very handy.

```
// test vaues
$currentfilename="inThisFile.scad";
$mainfilename="topLevelFile.scad";
$linenumber=1234; // mockup
function now(format) = format[0]=="Y" ? "20190205" : "155539"; // mockup

module imprint(line=0) {
  date=now(format="YYYYMMMDD");
  time=now("HHMMSS");
  what=$mainfilename;
  this=$currentfilename; 
  stampmain=     [
                  str("Built: ", date, " @ ", time)
                , str("   By: ", what)
                 ];
  stampcurrent=  [
                  str("  Via: ", this, line!=0 
                                            ? str(": ", line)
                                            : "")
                 ];
  lines= this!=what ? concat(stampmain, stampcurrent) : stampmain;

  lineoffset=20;
  numlines=len(lines);
  mirror([0,1,0])
    for (line=[0:numlines-1])
      translate([5,-(line+1)*lineoffset])
        text(lines[line]);
    
}  

module emboss(deep=3, offset=[0,0]) {
  difference() {
    children(); 
    linear_extrude(height=deep)
      translate(offset)
        imprint($thisline);
  }
}

module part() cube([200,100,10]);

module somecomplexstuff() {
  // ...
  // make the thing
  $thisline=$linenumber; 
  emboss()
    part();
  // ...
 // other parts
}

somecomplexstuff();
```

MichaelPFrey, 2019-02-10:
for those that want to see an possible implementation as a base: #2741 #2633

see also #380 for an older discussion

MichaelPFrey, 2019-02-14:
After a lot of thinking, I can now much articulate one of my main concern: Usually, I do not not want/need to know on which line and file I am currently am. I could use a random string to indicate that. Much more interesting is, who is calling the module or function. This would/could lead to the antipattern of `myModule(..., $linenumber, $currentfile)`.  We do not have a preprocessor, so there is no elegant way to hide that in user space.

So a specialvariable like $callerlinenumber and $callerfile would be useful and avoid that antipattern. That would also allow very powerful modules:
```
module dbg(){
   echo($callerlinenumber);
   echo($callerfile);
}
```
Beeing consequent, those special variables might as well be vectors, to prevent people from building a user space call stack tracer.

Currently, that is just an idea and I do not want to go that route, when it means eating up RAM.

jordanbrown0, 2021-03-06:
If you want to figure out the call stack, it seems like what you want is something like `parent_module(n)` that returns the file name and line number for N levels up the stack.

jordanbrown0, 2025-03-09:
`$` variables aren't magic enough for current-line and current-file.  They aren't really magic at all, just different, call-scoped instead of lexically-scoped.  You'd need a different sort of token, perhaps one expanded by the lexer or the parser.

Information about the *caller* could come from $ variables, but I think expanding on `parent_module()` would be better.