openscad/openscad #2770: Object doesn't Display on Laptop with Dual-video

zBeeble42, 2019-02-08:
I have an HP zBook.  What's special about this laptop is that it has the regular 'i7' Intel video but it also has a Quadro k2100m (nVidia) graphics card.  On my workstation, everything works fine, but when I transfer the file to my laptop, only part of the object is rendered.

I've attached the object.  It isn't very complex.  I've also attached two screenshots.  One is the default view when I load.  The second shows the result after a tiny amount of rotation ... it's janky... things clip in an out like this.  The cylinders are actually part of difference statements and shouldn't ever show.

[FanJoint_scad.txt](https://github.com/openscad/openscad/files/2846636/FanJoint_scad.txt)
[image]
[image]

thehans, 2019-02-08:
Can you paste your results from Help -> "Library Info", to determine that it detecting the Nvidia card?
Possibly try a driver update.

Also if you are running a 2015.X release, try a more recent [development snapshot](http://www.openscad.org/downloads.html#snapshots) and see if that helps things any.

MichaelAtOz, 2019-02-08:
Also try Preferences/Advanced/Force-goldfeather

zBeeble42, 2019-02-08:
OpenSCAD Version: 2015.03-2
System information: Windows 8 (v6.2.0.0) 8 CPUs 31.43 GB RAM
Compiler: GCC "5.2.0" 64bit
Boost version: 1_57
Eigen version: 3.2.5
CGAL version, kernels: 4.5, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.5.1
QScintilla version: 2.8.4
MingW build: MingW64
GLib version: 2.44.1
Application Path: C:/Program Files/OpenSCAD
Documents Path: C:\Users\dgilbert\Documents
Resource Path: C:/Program Files/OpenSCAD
User Library Path: C:/Users/dgilbert/Documents/OpenSCAD/libraries
User Config Path: C:\Users\dgilbert\AppData\Local/OpenSCAD
Backup Path: C:/Users/dgilbert/Documents/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
C:/Users/dgilbert/Documents/OpenSCAD/libraries
C:/Program Files/OpenSCAD\libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.12.0
OpenGL Version: 4.3.0 - Build 20.19.15.4531
GL Renderer: Intel(R) HD Graphics 4600
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Using QGLWidget

GL Extensions:
GL_3DFX_texture_compression_FXT1
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
[... 161 lines cut ...]
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_ARB_buffer_region
WGL_ARB_create_context
WGL_ARB_create_context_profile
WGL_ARB_create_context_robustness
WGL_ARB_extensions_string
WGL_ARB_framebuffer_sRGB
WGL_ARB_make_current_read
WGL_ARB_multisample
WGL_ARB_pbuffer
WGL_ARB_pixel_format
WGL_ARB_pixel_format_float
WGL_EXT_create_context_es2_profile
WGL_EXT_create_context_es_profile
WGL_EXT_depth_float
WGL_EXT_extensions_string
WGL_EXT_pixel_format_packed_float
WGL_EXT_swap_control
WGL_EXT_swap_control_tear
WGL_INTEL_cl_sharing
WGL_NV_DX_interop
WGL_NV_DX_interop2

zBeeble42, 2019-02-08:
Turning on the goldfeather checkbox works.  It' also possible, I've found, to have nVidia force it to use the nVidia GPU (in the nVidia settings app).  What does this mean, then?

thehans, 2019-02-08:
As your Library Info shows, its defaulting to  `GL Renderer: Intel(R) HD Graphics 4600`
I think this is default on laptops for power saving reasons.  
There are some issues with some intel GPU rendering in OpenSCAD (I don't know the details) that the goldfeather(alternate rendering method) works around.

You should be able to set either the GPU per application, or globally for your whole system if wanted: 
 https://appuals.com/how-to-change-default-display-adapter-on-windows-7-8-10/
If you are running on battery it might drain faster though.  Also I believe this requires a reboot.

I would recommend using the nvidia which probably doesn't need goldfeather option enabled.

You can check OpenSCAD Library Info again after changing setting to verify its running on nvidia.

closed 2019-02-11 by thehans