openscad/openscad #2771: MXE build generates over 56,000 warnings

thehans, 2019-02-09:
I noticed the build logs from CircleCI are absurdly long, ~150MB, and they seem to be filled with nothing but bogus compiler warnings from `x86_64-w64-mingw32.static.posix-g++`

They're all about "optimization attribute"s not matching, and seem to be a warning for one thing, then goes on to show a "previous definition" from a completely different function signature.

Here an example of the first one of these warnings from https://circleci.com/gh/openscad/openscad/1601
```
In file included from ../src/primitives.cc:30:0:
../src/evalcontext.h:42:69: warning: optimization attribute on 'std::ostream& operator<<(std::ostream&, const EvalContext&)' follows definition but the attribute doesn't match [-Wattributes]
 std::ostream &operator<<(std::ostream &stream, const EvalContext &ec);
                                                                     ^
In file included from ../src/expression.h:7:0,
                 from ../src/exceptions.h:5,
                 from ../src/feature.h:9,
                 from ../src/module.h:5,
                 from ../src/primitives.cc:27:
../src/value.h:227:24: note: previous definition of 'std::ostream& operator<<(std::ostream&, const Value&)' was here
   friend std::ostream &operator<<(std::ostream &stream, const Value &value) {
                        ^~~~~~~~
```
and it also seems to throw one for every Qt function in existence.

Should this be reported to MinGW project?

thehans, 2019-02-09:
Similarly there are many warnings on the AppVeyor builds too, though not nearly as many as CircleCI.  The AppVeyor log is merely 3.3MB in size.

Both using MinGW, but AppVeyor is on msys2, and CircleCI on MXE.

thehans, 2019-02-10:
I spent some time narrowing down the cause and it seems that including Eigen/Core between different headers that define operators causes gcc to go crazy.  I've submitted a bug report to MinGW-w64 [here](https://sourceforge.net/p/mingw-w64/bugs/786/) with a simplified test case, and attempted to email Eigen team about it.

thehans, 2019-02-13:
The turns out its a gcc bug and not just on mingw, its just that Eigen has some conditional preprocessor setting that it only turns on in mingw environment, which is what triggers the issue.  Bug is now tracked on gcc project: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=89325

t-paul, 2019-11-11:
Workaround is merged.

closed 2019-11-11 by t-paul