openscad/openscad #2777: Add dpi= and unit= to SVG import()

kintel, 2019-02-10:
SVG doesn't contain unit information, and it's not common to export with millimeters as units, so we may need some extra parameter to help users get the correct dimensions in OpenSCAD.

Suggestion:
`import("mydesign.svg", dpi=72, units="mm");`

..would be equivalent to:
`scale(25.4) scale(1/72) import("mydesign.svg");`

FIXME:
* Upload a few Illustrator examples
* Upload a few Inkscape examples

nophead, 2019-02-10:
Confusing! Does it mean the SVG is units are 72 DPI and you want OpenSCAD units to be mm?

SVG files can have units in them. mm, px, dpi, etc.

Since going the other way OpenSCAD units are mm in exported SVG file I don't think we need the extra parameters.  Just assume OpenSCAD is in mm and obey the units in the SVG. If doesn't have any then make it 1:1 with OpenSCAD units, just as with STL.

kintel, 2019-02-10:
Confusing indeed.

I'm trying to work through this as a user workflow, starting with Illustrator as it's a commonly used tool. Illustrator exports SVG with implicit units of 72 DPI, and I'm trying to figure out how to give people tools they can use in practice. Power users of Illustrator can change the export DPI; we should probably test that as well, but getting it to output millimeters might be tricky.

Perhaps Inkscape is a better citizen - I haven't tried to create any structured designs in Inkscape yet..

nophead, 2019-02-10:
Yes SVG is a mess when used for laser cutting, etc, because it is designed for doing graphics which just fit an available space. People often have dimension problems without OpenSCAD being involved.  If OpenSCAD works in a dimensionless way unless the SVG includes units I think that is as complicated as it needs to get. People will learn they have to output with units, just as they do when using laser cutters because there isn't a standard DPI. If it has units then convert them to mm as that probably covers most peoples use case. If they are using inches then they probably have scale factors anyway.

kintel, 2019-02-10:
_"learn they have to output with units"_ being the issue when design software doesn't export units.

With laser cutting, this is usually hooked into the design tool, e.g. as a printer driver which manages units. We don't really have the luxury of writing exporters for design tools, so this is the best I can think of.

t-paul, 2019-02-10:
How does it work? It's assuming the file is in inch and the mm introduces the 25.4 scale factor? I'll check but I don't think that will work for Inkscape.

kintel, 2019-02-10:
The SVG I'm looking at (the one posted above) is in "dots". In export settings you can specify DPI (dots per inch) as that's the standard for paper printing. ..so to get inches we divide the file's unit by the DPI setting. To get millimeters, we multiply by 25.4.

I feel Inkscape SVG will have similar behavior, perhaps different per platform if we're lucky : /

t-paul, 2019-02-10:
Ah, ok. So maybe drop the `units="mm"` part for now as @nophead suggests?

nophead, 2019-02-10:
If the file has units obey those. If not you could specify a unit but it could be anything. Not sure it is always a DPI and certainly can by 96 as well as 72 and varies between programs and versions of programs. But I think with all of them you can set them up to output specified units, so I don't think OpenSCAD needs to do it.

t-paul, 2019-02-10:
I'm not sure if I'm reading the [Spec](https://www.w3.org/TR/SVG11/coords.html#Units) correctly, but it seems fully compliant that we define the user unit to be 1 in OpenSCAD. That would map both unit-less and `px` values 1:1. Everything else would depend on the DPI setting (making it relatable to real word dimensions). But I think I have to read that a couple of more times :smile: .

kintel, 2019-02-10:
Yep, _default_ behavior should probably be to treat 1 SVG unit as 1 OpenSCAD unit.

I'd love to see someone export SVG with specified units from Illustrator. That would make this issue much less important.

kintel, 2019-02-25:
Fixed by #2775

closed 2019-02-25 by kintel