openscad/openscad #2781: Spacemouse - hidapi - Preference window blank on 2nd visit

MichaelAtOz, 2019-02-11:
I just installed 2019.01.RC2 & 2019.02.05.ci1524 to test with my SpaceMouse (wireless).
The axes were wrong so I went to Preferences.
a. Is it expected behaviour that the preview window does not respond to inputs when the Preferences window is open? Noting the coloured bars in Preferences do respond.
b. After closing Preferences, then opening it again I get a blank window.

[image]

If I right click I get a blank option box, if I click on it I get the top line of the Preferences window (3D View/Editor/Features etc), but clicking any of them still has the bottom section blank.
I have to close/reopen to get it bank (for the one time only).
Windows 7/64.

t-paul, 2019-02-11:
Yes a) is expected. It's also part of the logic that tracks the currently active window. As device input is global regardless of the currently selected application (at least in some cases), we stop handling events once the window is not active.
I've never seen the b) issue on Linux, does it come back when doing things like moving, hiding/showing or resizing any or the OpenSCAD windows? Is everything else still working?

MichaelAtOz, 2019-02-12:
I've changed the title as I'm not sure it is SpaceMouse related, as it occurs if I don't visit the Axis/Buttons tabs.

> does it come back when doing things like moving, hiding/showing or resizing any or the OpenSCAD windows? 

No. As I said, the only change is if I right click the panel it shows a blank option box, clicking it restores the list of tabs (3D View/Editor/etc), but nothing underneath any, right click again shows the same small option box, but with a tick; unfortunately it disappears before I PrintScr. If I close it and Edit/Preferences again the list of tabs is gone, totally blank again.

>Is everything else still working?

I was about to say yes, Including SpaceMouse, Help/About/Font-list/Library, however
I went back to Library-info and it is blank, as is font-list.
Restarting, Library-info, n-times is OK, then font-list once is OK, then both come up blank.
Same if reverse the order.
Interestingly, after the above, ie when blank, Preferences is still OK once, then blank 2nd time.

Library info from RC2:
[blank library info.txt](https://github.com/openscad/openscad/files/2853634/blank.library.info.txt)

Does the `*` in `InputDrivers: HidApiInputDriver (256f:c62e - 3Dconnexion Space Mouse Wireless (cabled))*` mean anything?

I'm now going to download some older snaps to see when it started.

MichaelAtOz, 2019-02-12:
@nophead Can you give [RC2](http://www.openscad.org/downloads.html#snapshots) a go to check if the above can be replicated. 
I was about to announce RC2 on the forum, but will wait if this is confirmed.

MichaelAtOz, 2019-02-12:
Hmmm, this does NOT happen on my normal box (currently running as a VM via Windows RDP), which does not have the SpaceMouse connected...

MichaelAtOz, 2019-02-12:
64bit unless indicated otherwise '(32)'. 
Normal: 2015.03-2(32), 2017.10.28*, 2018.05.22, 2018.05.24, 2018.05.30, 2018.08.12, 2018.08.13#, 
Blanks: 2019.01-RC2*, 2019.01.05.ci1524*, 2018.11.20.ci571*, 2018.09.05, 

`*` indicates SpaceMouse supported version.
`#` shows version as 2018.08.12(git bbc99b8)

So change is between 2018.08.13 & 2018.09.05.

Note there is 'OpenSCAD-2018.08.18-x86-64_mxe-gcc7.zip' but I didn't try that as I don't know what the gcc7 bit is about. (see below - that was when gcc7 was introduced)

Library compare:
[image]

nophead, 2019-02-12:
The preferences window works for me but RC2 doesn't include is_list(), so doesn't work with my code.

nophead, 2019-02-12:
Given it reports Git version 35858ead how does one find that commit?

nophead, 2019-02-12:
Ha, it replaced the version with a link in my post so that goes to it but how do you get to it without posting a link?

kintel, 2019-02-12:
you make your own link: `https://github.com/openscad/openscad/commit/35858ea`
..or ask git:
```
git checkout 35858ea
git log -p 35858ea
```

nophead, 2019-02-12:
Thanks. I didn't realise the short version would work in a URL as the link it created used the long version.

MichaelAtOz, 2019-02-13:
Hmmm, this does NOT happen on the Windows 7 box hosting the RDP session with SpaceMouse connected.

So it may be graphics card related??
As above, the box with the blanking problem is NVIDIA GeForce GTX 750Ti (with Intel integrated on board but not used),
it  does not happen via RDP which reports as:
```
GLEW version: 1.12.0
OpenGL Version: 1.1.0
GL Renderer: GDI Generic
GL Vendor: Microsoft Corporation
RGBA(8888), depth(16), stencil(8)
GL_ARB_framebuffer_object: no
GL_EXT_framebuffer_object: no
GL_EXT_packed_depth_stencil: no

Qt graphics widget: QGLWidgetGL Extensions:GL_EXT_bgra
GL_EXT_paletted_texture
GL_WIN_swap_hint
```

or on my RDP hosting NVIDIA Ion/integrated graphics:
```
GLEW version: 1.12.0
OpenGL Version: 3.3.0
GL Renderer: ION/integrated/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QGLWidgetGL Extensions:
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
<a long list more - let me know if you want to see this>
```
This may help re what works or not:
[image]

MichaelAtOz, 2019-02-25:
```
		Works		Doesn't
OpenSCAD	2018.08.12	2018.09.05
Compiler	GCC 4.9.4	GCC 7.3.0
CGAL Version	4.6.3		4.11
Qt version	5.7.0		5.11.1
QScintilla ver.	2.8.4		2.10.4
GLib version	2.44.1		2.50.2
```
As this appears to be a Qt windowing issue I'm presuming CGAL & QScintilla can be ignored.
I presume GLib is too low level to be likely.
So GCC or Qt.
Does this [Qt 5.11/12 bug](https://bugreports.qt.io/browse/QTBUG-67903) fit?

kintel, 2019-02-25:
We don't use QML, so that's probably not it.