openscad/openscad #2784: Crash from misc/escape-test.scad

thehans, 2019-02-12:
I noticed this from manually trying out some of the tests as I was refactoring for #2783.

It only happens if you load the file `testdata/scad/misc/escape-test.scad`, hit F5, then F6.  Unfortunately there isn't any test script that is made to duplicate this exact case.

OpenSCAD exits and I see this output on the command line:
```
openscad: src/CGALRenderer.cc:48: CGALRenderer::CGALRenderer(std::shared_ptr<const Geometry>): Assertion `ps->getDimension() == 3' failed.
Aborted
```

allenbangai, 2019-02-15:
Hi, my name Kamadje Allen and I code in C++. I will llike to contribute to you organisation.
Please, may I know the if contributing to this issue be considered for GSOC.?
@thehans

kintel, 2019-02-15:
@allenbangai We'd love to see GSoC applicants interact with the community and write some code, as that's a great way of evaluating a candidate, so yes that would be very beneficial.

@thehans Please sync with Kamadje so you don't both work on this at the same time.

thehans, 2019-02-15:
No problem, I haven't attempted to debug this particular issue yet.  

I haven't been involved in GSoC projects before, but I think this one would be fairly minor in terms of code required to fix it, probably just adding an extra check somewhere.  Also in terms of overall impact, its a very niche case that I doubt comes up in regular use.  That said, it might be a good low-hanging intro to the code base, and any contributions are appreciated :)

kintel, 2019-02-15:
@thehans This would just be a chance to let a candidate demonstrate skills as part of the application process.

thehans, 2019-02-15:
OK, sounds good then.

allenbangai, 2019-02-15:
Thanks very much. I am Kamadje.

Provenance : Courrier pour Windows 10

De : Marius Kintel
Envoyé le :vendredi 15 février 2019 19:17
À : openscad/openscad
Cc : Kamadje Allen; Mention
Objet :Re: [openscad/openscad] Crash from misc/escape-test.scad (#2784)

@allenbangai We'd love to see GSoC applicants interact with the community and write some code, as that's a great way of evaluating a candidate, so yes that would be very beneficial.
@thehans Please sync with Kamadje so you don't both work on this at the same time.
—

---
L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

commit 355aaddd (2019-03-31): Fixes #2784.
commit 9877f19b (2019-03-31): Merge pull request #2906 from nophead/Fix_2784

Fixes #2784.
closed 2019-03-31 by kintel