openscad/openscad #2786: Issues with SpaceMouce (etc) support

MichaelAtOz, 2019-02-12:
I have a few issues.

First pedantics, [Axis v's Axes](https://grammarist.com/homophones/axis-vs-axes/), should the preferences tab be Axes and the label 'Axes Mapping' & button 'Auto Trim Axes'?

Actual bugs to follow.

MichaelAtOz, 2019-02-13:
Not sure of the terminology, so 'Axis n' describes the detection of movement of the control knob, as reflected by the coloured bars in Preferences. I'll use 'Channel' to describe the pairing of the X,Y,Z columns with the rows (Translation/Viewport-rel-Translation/Rotation etc), the result of which causes the preview to do something. Let me know if there is a more appropriate term.
So the Channel Y+Translation moves the objects in the preview up/down when Mapped to an Axis.

There is a problem with at least one Channel, I have yet to try all Mappings, which I'll get to.

Y+Viewport-rel-translation (VRT) Channel is not responding to input.
When mapped to Axis-1, with Preferences open beside the Preview, and the Main/Preview Window active, I can see the coloured bar for Axis-1 change, but there is no change in the Preview.  
I have X+VRT mapped to Axis-0, which works, moving the view left/right.

If I reverse this, X-VRT=Axis-1, Y-VRT=Axis-0, then up/down on the control-knob moves left/right, but left/right on the control-knob does not move.

There is also seems to be a difference in responsiveness on some axes, I'll have to try to quantify that...

MichaelAtOz, 2019-02-13:
This is with HIDAPI on Windows 7/64 BTW, with 2019.01.05.ci1524.

MichaelPFrey, 2019-02-13:
> Y+Viewport-rel-translation (VRT) Channel is not responding to input.

You are in orthogonal view. Please change it to perspective to see what it does. Or look in the bottom left corner, where translate = changes. This is not a bug, this is a very specific feature. When you map zoom to one axis and Y+Viewport-rel-translation to an other while in perspective view, you should get the vertigo effect.
Most users will therefor use zoom as it works in both orthogonal and perspective. Before you wonder why you can map two axis to zoom: Game controllers have two shoulder buttons.

Yes, this has little to no real world use, but the inputdriver is about giving as much control to the user as possible.

MichaelPFrey, 2019-02-13:
Maybe I also need to add, that openscad internally considers Z tobe up and down.
X is left/right. Y is "into and outof the screen".

MichaelAtOz, 2019-02-13:
I'll reexamine the above.
Next issue.
I have had the one process running all day.
I have changed various mappings and a few button settings.
Now buttons are not responding.
They were set to various view-> settings, which were working earlier.
I have changed button 0 & 1 to Help/Library-info & Help/About.
The preview is responding to movements, but there is no response to buttons.

Is there a command line --debug arg which may help with this?

MichaelAtOz, 2019-02-13:
I'm talking about SpaceMouse buttons.

MichaelAtOz, 2019-02-14:
The button issue is when I hide the editor.
To reproduce, 
Main window with Editor, Console & Preview
Edit Preferences, position beside Main window (I leave this open due to the other blanking issue)
Click Main window, buttons work
Hide Editor & Console, buttons work
Click on Preferences window
Click back on Preview, buttons no longer work.

However if I close Preferences, it works.
But if I then close the Preview (ie app), rerun it (it remembers editor & console are hidden), load the previous file, F5, the buttons no longer work.
I think the window with the menu needs to grab focus?

MichaelAtOz, 2019-02-14:
If you unhide the Editor, you actually need to click in the edit pain to get buttons back.

MichaelAtOz, 2019-02-14:
@MichaelPFrey What does the Gain control do?

MichaelAtOz, 2019-02-14:
> There is also seems to be a difference in responsiveness on some axes, I'll have to try to quantify that...

Setup:
Code for test object
```
color("red") cube([5,1,1],true);
color("springgreen") cube([1,5,1],true);
color("blue") cube([1,1,5],true);
```
Hide the console.
Size the preview so the viewport is square
[image]
Edit/Preferences & position next to preview so you can see the bar graphs.
Setup buttons & mappings like so:
[image]
Press button 0 then 1, you should have something like so:
[image]
You can press buttons 0 then 1 again to reset the view.
Now use the SpaceMouse to translate left-right at full lock, checking the Preferences axis bars to check the deflection. Reset the view and do up-down.
If you get what I do, then X (left right) will be sluggish, but Y (up/down) is nippy.

MichaelPFrey, 2019-02-14:
> @MichaelPFrey What does the Gain control do?

you can call it sensitivity. Higher value = higher sensitivity. Lower value = lower sensitivity. The sensitivity of the zoom is the most subjective/depending on the input device. The shoulder triggers of a game controller idle on -1, so they need an offset of 1, making the max. value returned 2, which is overly fast. With a 3D mice on the other-hand, I would increase the gain, as otherwise the zoom is really slow. 

Interestingly,  the non linear scaling is done before the gain is applied.

https://github.com/openscad/openscad/blob/ac1d89f2381f8d4b6eaf0492c8ba1666d7d7f393/src/input/InputEventMapper.cc#L113
https://github.com/openscad/openscad/blob/ac1d89f2381f8d4b6eaf0492c8ba1666d7d7f393/src/input/InputEventMapper.cc#L97
https://github.com/openscad/openscad/blob/ac1d89f2381f8d4b6eaf0492c8ba1666d7d7f393/src/input/InputEventMapper.cc#L106
https://github.com/openscad/openscad/blob/ac1d89f2381f8d4b6eaf0492c8ba1666d7d7f393/src/input/InputEventMapper.cc#L90-L95

MichaelAtOz, 2019-02-15:
> If you get what I do, then X (left right) will be sluggish, but Y (up/down) is nippy.

Perhaps it needs Gain per Axis.

MichaelPFrey, 2019-02-18:
> Perhaps it needs Gain per Axis.

Absolutely valid - I just have to figure out how to design the UI and what variable I actually want to effect and how. And then I need to hope that users understand how to use all those controls.

MichaelAtOz, 2019-02-19:
Is it possible to change the type of the bar graphs in Preferences?
They appear to be progress bars, with Aero they have this anoying, to me anyway, throbbing effect, presumably to imply progress (when non exists...).
See screen recording: [clip0002.zip](https://github.com/openscad/openscad/files/2878930/clip0002.zip)
Note how some throb and it changes after moving the SpaceMouse. No idea what that is meant to represent in a UI.

MichaelAtOz, 2019-02-19:
> how to design the UI

[image]

?

t-paul, 2019-11-14:
Sorry, but this is getting too confusing. Some parts are done, others are more discussion. I'll close this issue, if there are still open bug/issues can you please move those to a new issue?

closed 2019-11-14 by t-paul