openscad/openscad #2787: stack usage for module calls

MichaelPFrey, 2019-02-15:
Test case:
```
module test(x){
    if(x>0){
        test(x-1);
    }
}
for(a=[1:10]){
    z=a*1000;
    echo(z);
    test(a*1000);
    echo("ok");
}
```
On Ubuntu 18.10 build with current master:
```
ECHO: 1000
ECHO: "ok"
ECHO: 2000
TRACE: called by 'test', in file recdetect.scad, line 3.
```
with  OpenSCAD 2019.01.23 "nightly"
```
ECHO: 1000
ECHO: "ok"
ECHO: 2000
ECHO: "ok"
ECHO: 3000
ERROR: Recursion detected calling module 'test' in file michael, line 1
```
On Ubuntu 18.10 build on the 2019.01.17 (I have no idea what this build truly is, most likely it was the feature branch I was working on on that day)
```
Compiling design (CSG Tree generation)...
ECHO: 1000
ECHO: "ok"
ECHO: 2000
ECHO: "ok"
ECHO: 3000
ERROR: Recursion detected calling module 'test' in file recdetect.scad, line 1
```
Note that you can not compare windows to linux, as the build infrastructur and stackusage betwhen windows and linux are different. I also should not compare my custom build against a nigthly build as it uses a different build infrastructur and runtime libraries

This goes back to https://github.com/openscad/openscad/issues/1443#issuecomment-462171597

Testing around, I suspect #2645
https://github.com/openscad/openscad/blob/ac1d89f2381f8d4b6eaf0492c8ba1666d7d7f393/src/UserModule.cc#L43-L44
https://github.com/openscad/openscad/blob/ac1d89f2381f8d4b6eaf0492c8ba1666d7d7f393/src/ModuleInstantiation.cc#L86

commenting out those lines, seams to do the trick. Off-course I do not want to do that. Luckly, we have a well documented pattern to solve that kind of issue:
https://github.com/openscad/openscad/blob/ac1d89f2381f8d4b6eaf0492c8ba1666d7d7f393/src/context.cc#L182-L194
I will work on a more formal test and then a fix during the weekend.

MichaelPFrey, 2019-02-22:
There is/was discussion on putting the whole evaluation in a separate thread.
That should as a side effect significantly increase the amount of stack we have during this phase.

jamessynge, 2019-02-24:
What kind of thread does the evaluation happen on currently? A UI/event loop thread?

kintel, 2019-02-24:
@jamessynge The evaluation currently happens in the main thread, which is also the UI/event loop thread.

t-paul, 2019-11-14:
fixed via #2788

closed 2019-11-14 by t-paul