openscad/openscad #2791: OS X 10.14.2 homebrew build issue

ScaredyCat, 2019-02-16:
Mojave 10.14.2

Unable to build dependancies

kintel, 2019-02-19:
@t-paul I wonder if this is due to Ruby not managing to unzip the lib3mf package due to unicode filenames?
When unzipping manually I get 
```
error:  cannot create lib3mf-openscad-build/TestFiles/Mixed/Box_+?????????.3mf
        Illegal byte sequence
```

t-paul, 2019-02-19:
Yes, that rings a bell. I seem to remember a pretty similar discussion on IRC (with @tcurdt ???). I'm not sure what the resolution was but maybe updating homebrew was involved.

tcurdt, 2019-02-19:
Yup, that was me. I've seen the exact same thing.
Unfortunately I cannot remember how I got past it.

t-paul, 2019-02-19:
I found the link :-)

Looks like replacing `unzip` was the key and this was the reference: https://github.com/adamhathcock/sharpcompress/issues/315

kintel, 2019-02-19:
Right, except homebrew does appear to use `ditto` already, so feels like a different issue..

t-paul, 2019-02-19:
Maybe new versions of homebrew? The error is *exactly* the same as the one from @tcurdt : https://gist.github.com/tcurdt/128d1b6d86fd8a635e678afbeb496273 - only one of the line number from the stack trace differ by one line.

kintel, 2019-02-20:
@t-paul the homebrew formula pulls from your github - can we use the official one now that they have a more recent release out?

Also, a proper fix appears to be to use the tarball rather than the zip.

t-paul, 2019-02-20:
Right, using the release version from the main repo would make sense. I'm not sure I can test though. My MacBook I hope to use for MacOS stuff is still lying around as a heap of parts as I'm trying to remove the totally puffed up battery :sob: .

ScaredyCat, 2019-02-20:
Is there anything I can do here to help with the testing ?

kintel, 2019-03-09:
@ScaredyCat Try now - I've updated the lib3mf package

ScaredyCat, 2019-03-10:
Yes, this fixes the problem, (although I have another now ;) ), lib3mf now installs via homebrew correctly. Thank you.

closed 2019-03-10 by ScaredyCat