openscad/openscad #2794: Editor not handling Unicode properly

MichaelAtOz, 2019-02-17:
While looking at Windows unicode in #2658, I came across this.
I'm raising it separately as it is not related to the filesystem, and I suspect it may not be limited to Windows. (and it may take me some time to finish that exercise)

2019.01-RC2 on Windows7/64.

The following code from [here](https://github.com/openscad/openscad/issues/512#issuecomment-28349445), by implication was displayed correctly at that time,
```
texts_array = [
"DEADBEEF",
"Ленивый рыжий кот",
"كسول الزنجبيل القط",
"懶惰的姜貓",
"äöü ÄÖÜ ß",
//Removed for github comments - Was unicode above 0xffff
"⠁⠂⠃⠄⠅⠆⠇⠈⠉⠊⠋⠌⠍⠎⠏",
//Removed for github comments - Was unicode above 0xffff
];

for (text_array_idx = [0:(len(texts_array)-1)])
{
        echo( "Array element=", text_array_idx, "for", texts_array[text_array_idx], " of len=", len(texts_array[text_array_idx]), "[0]=", texts_array[text_array_idx][0],":" );

for (text_idx = [0:(len(texts_array[text_array_idx])-1)])
{
        echo( " [", text_idx, ,"]=", texts_array[text_array_idx][text_idx] );
}
}
```

When pasted into the editor has two issues.
Firstly, on 3/3 Windows boxes, some is not displayed in the editor, but F5 echo output shows it is there.
[image]

Secondly, on one (the RDP) of three Windows boxes it causes the GUI to become unresponsive to mouse clicks, except in the editor. So the preview does not respond to drags, the menus/buttons do not react, can't highlight text in the console. Clicking another window and clicking back (outside the editor) then works until clicking inside the editor.

In the next post I'll also highlight some other Unicode examples.

MichaelAtOz, 2019-02-17:
This is a non-RDP box.

Note 2018 version.
[image]
Same on RDP box with RC2.

RC2
[image]

MichaelAtOz, 2019-02-17:
For reference,
* Windows RDP box is the source of the above DEADBEEF
* Windows Daul box is the above non-RDP box
* Windows Quad box (also non-RDP) is the following DEADBEEF [with added text() output]

[image]

Note it is a different editor font failure than above.

Also pasted the text, correctly, into Windows Wordpad
[image]

It is not specific to Truetype or Opentype.

MichaelAtOz, 2019-02-17:
Oops - didn't notice that braille  didn't work in Wordpad.

MichaelAtOz, 2019-02-17:
Here is a one I prepared earlier, for another issue, explaining the three boxes.
[image]

MichaelAtOz, 2019-02-17:
Hmm, that "OPENSCAD" example above, may be OK as they are just latin chars.
When font= is specified text() outout is OK.
[image]

MichaelAtOz, 2019-02-17:
But if I do the same paste into Wordpad:
[image]

Something not right...

thehans, 2019-02-17:
Wordpad is rich text, it carries some font info over through clipboard somehow.  I wouldn't expect OpenSCAD string literals or echo to retain any font like that.

thehans, 2019-02-17:
By the way I'm looking into fixing unicode support on Windows.  

I've found this: 
https://sourceforge.net/p/mingw-w64/wiki2/Unicode%20apps/
Which says we need to define `UNICODE`, `_UNICODE` and pass linker option `-municode`
The first two are already automatically included by qmake.  

The `-municode` linker option is a little trickier, since it requires changing our `int main (int argc, char **argv)` over to `int wmain (int argc, wchar_t **argv)`
So anywhere we use argv needs to be converted to also support wchar_t (conditionally when on Windows).  I'm still working on it.

One thing I found as a workaround which actually passed the non-ascii test case `\testdata\scad\misc\sfære.scad` was this:
1) Open Control Panel 
2) Go to "Region" settings
3) Select "Administrative" tab 
4) Under section "Language for non-Unicode programs", click "Change System Locale..."
5) Check the box "Beta: Use Unicode UTF-8 for worldwide language support" 
It should prompt you to restart windows.  I think this fixed a lot, maybe all of these sorts of issues.

This is on Windows 10 for me, I think earlier version have similar setting but maybe slightly different labeling.

I have changed mine back to unchecked for testing/development purposes.  I think if I can get the `int wmain(...)` stuff working then windows won't consider OpenSCAD to be a "non-Unicode program" anymore.

MichaelAtOz, 2019-02-17:
> Edit: Try pasting your font stuff into notepad instead ;-)

For the DEADBEEF it does it correctly
[image]

MichaelAtOz, 2019-02-21:
> Check the box "Beta: Use Unicode UTF-8 for worldwide language support"

I suspect that is now this is historical, buy FYI that is not present on Windows 7.

masgart, 2021-10-04:
> When pasted into the editor has two issues.
> Firstly, on 3/3 Windows boxes, some is not displayed in the editor, but F5 echo output shows it is there.

This problem with the missing Arabic text is not restricted to Windows, it also appear on my Linux setup (Ubuntu 20.04).
Just played around with this a bit using various editors and it seems to be a problem with the Scintilla-Editor.
- QtCreator (Qt-based, of course) displays the missing text just like the link it came from (on left side).
- gedit even went one step further and displayed the Arabic line on the right side of the editor (which assumes to me it is right-to-left text)
- another QScintilla-based editor (some python script using QsciScintilla) showed exactly the same problems and behavior as the openscad

Special behavior on my system identical between openscad and another QScintilla means:
- after pasting the text seems to be missing
- But when marking the text from the beginning of the seemingly empty line character by character (via shift + right-arrow) the text starts to appear character by character (hardly readable in white on light grey on my system)
- When marking the text the other way round starting from the line after (via shift + left-arrow) the complete text suddenly pops up after the 3rd character marked (when the cursor reaches the start of the empty line press left-arrow once more)

According to the Scintilla website there might well be problems with arabic characters:
_Scintilla provides only limited experimental support on Windows for right-to-left languages like Arabic and Hebrew. While text in these languages may appear correct, interaction with this text may not work correctly as occurs with other editors._