openscad/openscad #2795: Unicode issue on search selection

pincr, 2019-02-17:
Minor problem when highlighting the search text. Every national char (more bytes in UTF-8 encoding?) makes shift on  highlighting the search text:

[image]

It is issue only on highlighting. Previous, next and replacement works fine.

Test example:

```openscad
module test() {
	sphere(10);
}

// Text with ony ASCII characters
translate([10,-10,0]) test();

// Text with national char: test "Č" test
translate([0,10,-10]) test();

// Another text with national char: test "Ř" test
translate([-10,0,10]) test();

text("Jet another text with national char: test 'Ů' test");

translate([-10,-10,-10])
test();
```
OpenSCAD version 2019.02.16.nightly (git 38f3279)
Fedora 28 (64-bit only) - https://download.opensuse.org/repositories/home:/t-paul/Fedora_28/

MisterAwesome23, 2019-06-05:
I would like to fix this GUI bug. Can someone help me get started with the codebase. I'm a new member :)

kintel, 2019-06-05:
A great start would be to build OpenSCAD from source and reproduce this using your self-built binary. The join the IRC channel to get advice going from there.

MisterAwesome23, 2019-06-05:
Okay. I'll set up self-built binary and reproduce the bug. I'll be updated on IRC as soon as I set it all up. Thanks

t-paul, 2019-09-12:
Fixed via #3071.

closed 2019-09-12 by t-paul