openscad/openscad #2798: Support for writing tests of library functions: expect assertion

jamessynge, 2019-02-17:
I'm writing a file with some library functions, and would like to be able to write some tests in the same file, including expecting assertions to fail. In the terminology of [Google Test](https://github.com/google/googletest), I'd like to write a Death Test, where the invoked module or function is expected to fail, and the Death Test fails if the invoked module or function does not fail. This might look like:
```
module good_module(v) { assert(v, "This is bad!"); }
// Expected to assert, producing a message including
// the sub-string "This is bad". Since that is what
// happens, it runs without printing the expected
// assert, and doesn't stop eval:
assert_asserts("This is bad") {
  good_module(false);
}
// Expected to assert, producing a message including
// the sub-string "This is good". Since that does NOT
// happen, it prints the stack trace of the assert that
// does fail (i.e. with the "This is bad!" message),
// and stops eval.
assert_asserts("This is good") {
  good_module(false);
}
// Expected to assert, with any message, but no
// assert fails, so this assert_asserts fails and stops
// eval.
assert_asserts() {
  good_module(true);
}
```

t-paul, 2019-02-17:
Better support for writing libraries is certainly something which is needed and test suites are a big part of that. I'm not yet sure how to fit that into the model description itself. An OpenSCAD script is not an executable file/program, it's a description of a 3D geometry, so adding processing logic feels a bit strange. Maybe it would be an option to have some kind of external test harness that could poke at the model (maybe a bit similar to how testbenches for VHDL work? or how the current animation support which is calling the same model with a changing `$t` input parameter).

jamessynge, 2019-02-18:
I was wondering about that, especially the suppressing of the 3D geometry that might result from evaluating a module. OTOH, I find it works better to keep tests closer to the code (script) that is being tested, else they tend to drift apart.

I suppose an option would be to add another special variable, $testing, that activates such code? Or a keyword that identifies a block of such code, so that the parser/interpreter can identify that a block of top-level code in a file contains tests rather than geometry. That would allow OpenSCAD to (based on a preference?) automatically run such tests each time the file containing the tests is loaded or modified and reloaded, rather than only if that file is directly opened by the user.

```
$testing_block(expect_death=false) {
  rs = rands(0,1,1);
  assert(is_list(rs));
  assert(len(rs) == 1);
  r = rs[0];
  assert(r >= 0);
  assert(r <= 1):
}
```

MichaelPFrey, 2019-02-18:
just throwing out an idea, without having an implementation to back it up: I think it should be possible to write a parser based try catch "module", that uses c++ try catch to catch the actual exception. The problem would not be implementing it, but accepting the implications.

An other idea would be not to use assert, but MyAssert (or any other name). For production, MyAssert would just simply call Assert. For testing, it would output a warning and just continue.

MichaelPFrey, 2019-02-18:
A special variable like $testings needs a very good definition, as complicated and intentionally sensitive code will build on it. Note that OpenSCAD users are very good in finding weird quirks and building complex code around it. So writing an definition is important, so that we do not need to change things latter.

Are you thinking of something among the line of $preview? Abstracting that, we could add menu option/keyboard short cut that does evaluate, but not create geometry, with for e.g. $evalonly or $nogeom set. (just bouncing ideas - a more specific variable name leaves less room for interpretation)

MichaelPFrey, 2019-02-18:
> Or a keyword that identifies a block of such code, so that the parser/interpreter can identify that a block of top-level code in a file contains tests rather than geometry.

A more concrete idea: we could use "magic module name" like
```
module $selftest(){
   //your test code here
}
```
and always call/evaluate that module before running anything else. But what would be the failing criteria? And I  fear strange side effects, like variables from the test persisting to the code run latter.

And why can't the first line of your code be calling selftest?

jamessynge, 2019-02-20:
I'm unclear on what you mean by "why can't the first line of your code be calling selftest?". Could you clarify?

RamiAwar, 2025-03-28:
I think in general OpenSCAD is a great tool because it brings the advantages of software architecture to 3D modelling. One great tool in software is testing, another is defining interfaces.

I was strongly drawn to OpenSCAD because I can see it accomplish this easily with what's available already. Right now I can already create 'interfaces' and tests for them by testing intersections, overlaps, highlighting things in red, etc. 

But it would be really cool if I can create a separate file that can be run with some cmdline option and the output would be accessible from the command line. Potentially with the name of the test that succeeded/failed. This way we can start introducing things like CI/CD and automations to OpenSCAD files.

Is this something easy to do? Or more on the difficult side? I've never contributed to OpenSCAD but willing to look into it if it's a beginner friendly issue.

RamiAwar, 2025-03-28:
I found this after a while, and this shows how to do basic tests!: https://learnxbyexample.com/openscad/testing-and-benchmarking/

I think the one thing still missing that would allow me to write some basic tests is an "intereference()" function that yields true or false if any non-zero intersection is detected between several objects. That would allow me to do all sorts of stuff testing interfaces between parts in a larger assembly.

I know that's not the main use of OpenSCAD today but I see so much potential in that for the reasons mentioned above.

jordanbrown0, 2025-03-28:
I agree with https://github.com/openscad/openscad/issues/2798#issuecomment-464834362 : it seems like it should be straightforward to have a builtin module that wraps a `try...catch` around its children, and throws if they don't throw.  Probably it would discard the geometry, if any, that they generate.

Once #4478 integrates it could be more general:  it could be a function that accepts a block as input, and returns what it throws (or undef).

Such a mechanism would be explicitly for testing; it would never be used in production, because in production you should never be throwing.

---

With respect to checking for interference between two objects, with #4478 an intersection and render() can be used to check for overlap.  But that seems like a separate topic, related only in that it's for testing OpenSCAD programs.