openscad/openscad #2802: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end()

richalt2, 2019-02-20:
This error occurs when I try to F6 render more than one object at a time.  Objects are separated in space and not intersecting.   Unfortunately the objects are fairly complex, created by the "turbine blades" module nSplines from https://www.thingiverse.com/thing:1208001/files

OpenSCAD version 2018.06.01 (git 4455a12) 
Here is the console output of an F6 render attempt:

Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: 
  /opt/mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_external_structure.h Line: 1152 
Geometries in cache: 20
Geometry cache size in bytes: 938600
CGAL Polyhedrons in cache: 1
CGAL cache size in bytes: 0
Total rendering time: 0 hours, 0 minutes, 0 seconds
Rendering finished.

F5 preview is always fine, with console output:
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 20
Geometry cache size in bytes: 938600
CGAL Polyhedrons in cache: 1
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized CSG tree has 2 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds

I attach a screen shot, showing two objects, which fails.  If I F6 one at a time, they are fine.

[image]

My code, calling the nSplines library is probably hard to make out, unless one studies that library.  It is pretty cool, for making nicely curved surfaces and solids!

What more information can I provide?
My workaround:  generate one object at a time, driven by a script, and meld them together in Adobe MeshMaker. 
Rich

// try nSplines

use <../../curves/nSplines/files/splines.scad>
use <../../curves/nSplines/files/Naca_sweep.scad>

LeafThick = 5.0; 
Aleaf = [ // z, p0x, p0y, p1x, p1y, p2x, p2y, p3x, p3y, thick

  [0, 2.3, 0.01,       0.53, -0.1,   0.33, -0.06,  0.0, -0.02,   5.0    ],  //base
  [4, 3.0, 0.147,     1.392, 0.21,  0.696, 0.03,  0.0, -0.002,  5.0    ],
 // [8, 3.1, 0,         1.52, -0.08,   0.84, 0.0,     0.0, -0.02,  5.0    ],
  [7, 3.98, 0.52,      2.6, -0.29,   2.88, 0.55,      0.0, 0.0,    5.0    ], //widest, uplift of edges
  [16,  3.5, -0.02,     2.01, -0.02,   1.17, -0.2,   0.0, -0.5,  5.0 ],
  [20,  1.98, 0.01,     1.5, -0.02,    0.74, -0.35,   0.0, -0.5,  5.0 ],
  [21,  0.88, -0.22,    0.59, -0.24,    0.4, -0.33,   0.0, -0.5,  5.0 ] ,
  [22,  0.141, -0.31,      0.107, -0.316,   0.058, -0.313,  0, -0.33, 5.0] 
  //[22,  0.033, -0.5,      0.22, -0.24,   0.2, -0.35,  0, -0.59, 5.0] 
]; 

LeafLenSurface = 22; 

Bsplines = nSpline(Aleaf, 8); //interpolates and creates intermediate data points

Leaf = Lgen_dat(Bsplines);

function leafBezpoly(p0x, p0y, p1x, p1y, p2x, p2y, p3x, p3y, thick) = concat(
     [ [1.1*p0x, -thick] ], 
     [ for (t=[0:0.05:1.01]) 
         let(   Bpoint=PointOnBezCubic3D([p0x, p0y], [p1x, p1y], [p2x, p2y], [p3x, p3y], t) )
          [Bpoint[0], Bpoint[1] ] ],
     [ for (t=[1.0:-0.05:0]) 
         let(   Bpoint=PointOnBezCubic3D([p0x, p0y], [p1x, p1y], [p2x, p2y], [p3x, p3y], t) )
          [-Bpoint[0], Bpoint[1] ] ],
     [ [-1.1* p0x, -thick]]
     

     
     );

 

     
function Lgen_dat(Splines) =
   [ for (i=[0:len(Splines)-1])
       let (crossSec = vec3D(
                     leafBezpoly(Splines[i][1], Splines[i][2], Splines[i][3], Splines[i][4], Splines[i][5], Splines[i][6], Splines[i][7], Splines[i][8], Splines[i][9])   )
            )
        T_(0,0, Splines[i][0], crossSec)  
   
   ]; 

   

function BEZ03(u) = pow((1-u), 3); 
function BEZ13(u) = 3*u*(pow((1-u),2)); 
function BEZ23(u) = 3*(pow(u,2))*(1-u); 
function BEZ33(u) = pow(u,3); 

function PointOnBezCubic3D(p0, p1, p2, p3, u) = [ 
        BEZ03(u)*p0[0]+BEZ13(u)*p1[0]+BEZ23(u)*p2[0]+BEZ33(u)*p3[0], 
        BEZ03(u)*p0[1]+BEZ13(u)*p1[1]+BEZ23(u)*p2[1]+BEZ33(u)*p3[1], 
        BEZ03(u)*p0[2]+BEZ13(u)*p1[2]+BEZ23(u)*p2[2]+BEZ33(u)*p3[2]]; 

module Oneleaf(length, layer_lift) {
   layer_lift_tab = [0, -11, -6, -2];  //layer_lift values 1, 2, or 3,
      //this is amount to angle the leaf upwards.
    
       rotate([0, layer_lift_tab[layer_lift],0])
       scale([length/LeafLenSurface, length/LeafLenSurface, 1.0]) 
         translate([0,0, LeafThick-3]) 
         rotate([90,0,90])
           sweep(Leaf );
           //sweep(Leaf, showslices=true);

}
//translate([0,0, -3]) 
//Oneleaf( 22, 1); 

LeafCircleTable = [  // degrees from x, layer (1 is top, 3 is bottom), len
   [6, 1, 10], [14, 2, 11], [19, 3, 11],
   [31, 1, 12], [43, 2, 9]

];

module LeafCirclebyTab(len, dia, startPetal, lastPetal) {
    //for (i = [0: len(LeafCircleTable)-1])
    for (i = [startPetal: lastPetal])
        rotate([0,0,  LeafCircleTable[i][0] ]) 
           translate([dia/2, 0, 0])
             Oneleaf(1.3*LeafCircleTable[i][2], LeafCircleTable[i][1]); 
    
}

LeafCirclebyTab(10, 23, 0, 0);   //any one works.  Two at once fail F6. 
LeafCirclebyTab(10, 23, 4, 4);

richalt2, 2019-02-20:
OpenSCAD version 2019.01-RC2 (git 35858ead),
same failure.

MichaelAtOz, 2019-02-20:
When you get CGAL errors with two objects but not one it is almost certainly broken geometry, usually from `polyhedron()`. Basically one object does not invoke CGAL, two causes a union() thus calling CGAL.
Confirm this, do one, then add a cube(1), if it errors that is the easy tell.
See this [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_model_showing_up_with_F5_but_not_F6?), the View/Thrown-together shows the problem.
If that doesn't show you anything post back & I'll load-up your design and have a look.

richalt2, 2019-02-20:
You are correct, with one of my objects, then add a cube(size=1); not intersecting, the same error occurs.
With the Thrown-together view, my entire object is purple, so this is bad, correct?
I will try to figure out why.

nophead, 2019-02-20:
If it is all purple it is inside out. I.e. the winding order of the faces
is wrong.

richalt2, 2019-02-20:
Hurray, I was able to figure out how to get the library to generate legit polyhedrons!   
The Thrown-together mode is great!  
Also nice to have some error message during Rendering which warned me about bad polyhendrons!!
Great advice MichaelAtOz.

t-paul, 2019-02-22:
Sounds like we can close this issue?

richalt2, 2019-02-22:
Definitely close! Nice to place a feature request someplace for a friendly "invalid polyhedrons" message!

t-paul, 2019-02-22:
Thanks. Yes, better warning/error for mesh errors is on the wish list for a long time (#191) but it's tricky to achieve.

closed 2019-02-22 by t-paul