openscad/openscad #2804: Syntax error not detected, silently ignoring some primitives

MoonCactus, 2019-02-21:
_OpenSCAD version 2019.01-RC2 (git 35858ea)_
Forgetting the semicolon after a primitive before another primitive produces no syntax error. Only the first primitive is generated and shown. I would expect the parser to fail as it makes debugging the typo much harder otherwise.
```
cube()        // misses the syntax error, only primitive shown
cylinder();   // does not show up
```
Here is a second example:
```
sphere()     // misses the syntax error, only primitive shown
cylinder()   // does not show up
cube();      // does not show up
```

t-paul, 2019-02-21:
The syntax is valid, so this can't be a parser error, however there could be an additional check which modules do actually evaluate child modules. That is covered by #596.

closed 2019-02-21 by t-paul

MoonCactus, 2019-02-21:
Oh, right! I should have thought twice about it from another angle, and looked for children-related issues in github indeed, not syntax-related!

PR #2806 (2019-02-23): warning, when a builtin module without support for it has children
fixes #2804, partially adresses #596 

Todo:
- add test case
- update changelog
- ...

In scope of this PR:
- builtin modules

Not in scope of this PR:
- user modules

[image]
[image]
[image]