openscad/openscad #2805: I can't open 2019.02.20 (macOS 10.11.6)

proto-user, 2019-02-21:
Suggested by Marius in the mailing-list, the output when launched from the command line is:
Sorry, "OpenSCAD" can not be run on this version of macOS. Qt requires macOS 10.12.0 or later, you have macOS 10.11.6.

Benjamin

t-paul, 2019-02-21:
Meanwhile you should be able to run 2019.01.30 (which is effectively 2019.02-RC2). This does run on my MacBookPro which has also 10.11.6:

[image]

proto-user, 2019-02-21:
Yes,
Thanks. Contrary to what I said this morning 2019.01.30 is fine.

kintel, 2019-02-21:
That makes more sense, as I revamped the dependencies for the 2019.02.20 build

kintel, 2019-02-22:
Turns out Qt doesn't support 10.11 any longer, as it's too old for them.
I'll see if I can patch the Qt build system a bit, but we won't guarantee good behavior as we'd be pushing the Qt envelope a bit.

kintel, 2019-02-23:
@proto-user Try OpenSCAD-2019.02.23

proto-user, 2019-02-23:
You're incredible! I can't count the times you've saved my computer from obsolescence. I know it can't last for ever.
You're the only ones to care about us. Thank you.

PR #2811 (2019-02-24): macOS: Downgrade Qt to 5.9.7 to improve backwards compatibility
Fixes #2805
closed 2019-02-24 by kintel