openscad/openscad #2807: https://www.openscad.org links don't work

ScaredyCat, 2019-02-21:


kintel, 2019-02-21:
https://www.openscad.org/ works for me. In what way doesn't that work for you?

ScaredyCat, 2019-02-21:
Sorry, was editing my post when you replied.. ^^^

kintel, 2019-02-21:
Probably broken HTML. See here for the actual html: https://github.com/openscad/openscad.github.com/blob/master/downloads.html

ScaredyCat, 2019-02-21:
It appears to only be affecting Google Chrome (Version 72.0.3626.109 (Official Build) (64-bit)) - Safari and Firefox work fine.

thehans, 2019-02-21:
For what its worth the URL fragments (`href="#blah"`) on openscad.org have never worked for me in chrome.  I suspect it may be due to malformed HTML tripping up the browser somehow.

There's 14 errors on the downloads page for example:
https://validator.w3.org/nu/?showsource=yes&doc=http%3A%2F%2Fwww.openscad.org%2Fdownloads.html

t-paul, 2019-02-22:
Hold on to your hats, I'll try moving the issue to the openscad.github.com repo...