openscad/openscad #2808: Build issue with glib.h on macOS 10.14.3

maccesch, 2019-02-23:
When running make I always get this message:

```
In file included from src/ModuleInstantiation.cc:2:
In file included from src/ModuleInstantiation.h:4:
In file included from src/localscope.h:3:
In file included from src/Assignment.h:6:
src/value.h:12:10: fatal error: 'glib.h' file not found
#include <glib.h>
         ^~~~~~~~
1 error generated.
make: *** [objects/src/ModuleInstantiation.o] Error 1
```

I have installed the dependencies with `./scripts/macosx-build-homebrew.sh`. qmake ran fine. And I tried reinstalling glib and relinking it but no luck. I checked and I have a file /usr/local/include/glib-2.0/glib.h

How can I fix this?

nophead, 2019-02-23:
I also get that if I open openscad.pro in qtcreator in MSYS2, I don't know if it is related.

kintel, 2019-02-23:
If you post the compiler cmd-line for the object file in question, we can try to see what is missing

thehans, 2019-02-23:
Try this and see if its any different:  `qmake OPENSCAD_LIBDIR=/usr/local`

maccesch, 2019-02-25:
@thehans Thanks. Unfortunately it doesn't make any difference.

@kintel Not sure what you need exactly. I hope this helps:

```
$ qmake OPENSCAD_LIBDIR=/usr/local CONFIG+=experimental
Info: creating stash file /Users/maccesch/Software/openscad/.qmake.stash
Project MESSAGE: Building experimental debug version
Project MESSAGE: Using ccache
Project MESSAGE: Using QOpenGLWidget
Project MESSAGE: Using libc++11
Project MESSAGE: Using C++11
```

Then after I run make

```
$ make
...
ccache /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -c -pipe -stdlib=libc++ -DSTACKSIZE=8388608 -stdlib=libc++ -std=c++11 -O2 -std=gnu++11  -arch x86_64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk -mmacosx-version-min=10.9 -Wall -W -Wno-overloaded-virtual -Wno-unused-parameter -Wno-unused-variable -Wno-unused-function -Wno-format-security -Wno-sign-compare -Wno-inconsistent-missing-override -Wno-unused-local-typedef -Wno-deprecated-register -fPIC -DOPENSCAD_VERSION=2019.02.25 -DOPENSCAD_SHORTVERSION=2019.02.25 -DOPENSCAD_YEAR=2019.0 -DOPENSCAD_MONTH=02.0 -DOPENSCAD_DAY=25.0 -DDEBUG -DSTACKSIZE=8388608 -DENABLE_EXPERIMENTAL -DENABLE_MDI -DUSE_QOPENGLWIDGET -DENABLE_QGAMEPAD -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_GAMEPAD_LIB -DQT_GUI_LIB -DQT_CONCURRENT_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -Isrc -Isrc/ext/libtess2/Include -I/usr/local/Cellar/qt/5.12.1/lib/QtWidgets.framework/Headers -I/usr/local/Cellar/qt/5.12.1/lib/QtMultimedia.framework/Headers -I/usr/local/Cellar/qt/5.12.1/lib/QtGamepad.framework/Headers -I/usr/local/Cellar/qt/5.12.1/lib/QtGui.framework/Headers -I/usr/local/Cellar/qt/5.12.1/lib/QtConcurrent.framework/Headers -I/usr/local/Cellar/qt/5.12.1/lib/QtNetwork.framework/Headers -I/usr/local/Cellar/qt/5.12.1/lib/QtCore.framework/Headers -Iobjects -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/AGL.framework/Headers/ -I/usr/local/include -I/usr/local/Cellar/qt/5.12.1/mkspecs/macx-clang -F/usr/local/Cellar/qt/5.12.1/lib -o objects/src/ModuleInstantiation.o src/ModuleInstantiation.cc
In file included from src/ModuleInstantiation.cc:2:
In file included from src/ModuleInstantiation.h:4:
In file included from src/localscope.h:3:
In file included from src/Assignment.h:6:
src/value.h:12:10: fatal error: 'glib.h' file not found
#include <glib.h>
         ^~~~~~~~
1 error generated.
make: *** [objects/src/ModuleInstantiation.o] Error 1
```

kintel, 2019-02-25:
Looks like glib-2.0 is indeed not located correctly.
What happens if you run this on the cmd-line?
```
pkg-config --cflags glib-2.0
```
..or perhaps you don't have pkg-config installed?

maccesch, 2019-02-25:
I get this output:
```
$ pkg-config --cflags glib-2.0
-I/usr/local/Cellar/glib/2.58.3/include/glib-2.0 -I/usr/local/Cellar/glib/2.58.3/lib/glib-2.0/include -I/usr/local/opt/gettext/include -I/usr/local/Cellar/pcre/8.42/include
```

kintel, 2019-02-25:
That's odd. Do you have any OpenSCAD-specific env. variables set? e.g. OPENSCAD_LIBRARIES?

maccesch, 2019-02-25:
No

kintel, 2019-02-25:
Best tip right now would be to mess around in this file to try to figure out what's going on:
https://github.com/openscad/openscad/blob/master/features/glib-2.0.prf

FYI, our CI uses `macosx-build-homebrew.sh` and works as expected, so somehow your environment ends up being different.

kintel, 2019-03-09:
@maccesch Any luck getting this to work on your system?

alan412, 2019-07-13:
fwiw, I am getting the same problem.   I am on 10.14.5 so I wonder if it is something to do with Mojave.

alan412, 2019-07-13:
I think I have figured out the problem, but it is bizarre.   It looks like even though they are being included on the CONFIG, that qmake isn't grabbing the feature files from the features/ directory...
(tested by putting message() in both the opens cad.pro at the end as well as in the pdf files in the feature directory)

monroewilliams, 2021-03-09:
I found that I had to add this to my command line variables to get it to find the features directory:
```
export QMAKEFEATURES=$OPENSCADDIR/features/
```

t-paul, 2021-03-09:
Note that `qmake` support will be removed soon, v2021.01 will be the last version using `qmake`.

monroewilliams, 2021-03-09:
I'm new to OpenSCAD development and was following the build instructions at https://github.com/openscad/openscad/blob/master/README.md , so I wasn't aware there was an alternative. What's the preferred new way to build?

t-paul, 2021-03-09:
All builds are currently migrating to `cmake` (#3652). The initial move is almost done, but there's still a number of open topics to solve (#2518).

kintel, 2022-01-02:
Since builds are now on cmake, this issue is likely resolved, or will resurface in a different incarnation. Please reopen if this pops up again.

closed 2022-01-02 by kintel