openscad/openscad #2813: Release prep: Test module caching

kintel, 2019-02-24:
Before the release:
* [ ] go through and verify all module cache tests at https://github.com/openscad/openscad/blob/master/testdata/modulecache-tests/README.txt
* [ ] Update/change tests as necessary.
* [ ] Pay special attention to error and warning that may interrupt compilation

nophead, 2019-02-27:
Test1: pass

Test2: module address does not change but nevertheless it picks up changes. I think deleting an object and creating a new one is likely to get the same address on the heap, so I don't think this test is valid.

Test3: pass

There is no Test4!

Test5: pass

Test6: pass

Test7: pass

Test8: pass

Test9: pass

Test10: pass

Test11: pass except the error marker in the editor doesn't go away unless F5 is pressed.  
#2821 fixes that.

Test12: pass same comment as Test11

Test13: passes with #2821 and stop on warnings turned off.

Test14: fails. 

When cascade2.sh is run:
```
Loaded design 'D:/msys64/home/ChrisP/openscad/testdata/modulecache-tests/cascadetest.scad'.
WARNING: Can't open include file 'cascade-B.scad'.
ERROR: Parser error in file "D:/msys64/home/ChrisP/openscad/testdata/modulecache-tests/cascadetest.scad", line 1: stop on first warning 
ERROR: Compilation failed!

WARNING: Can't open library 'cascade-D.scad'.
ERROR: Parser error in file "D:/msys64/home/ChrisP/openscad/testdata/modulecache-tests/cascadetest.scad", line 1: stop on first warning 
ERROR: Compilation failed!
```

The main file gets reloaded and compiled while the included files are missing, so lots of warnings. Without stop on warnings turned off I got this:

WARNING: Can't open include file 'cascade-A.scad'.
WARNING: Can't open include file 'cascade-B.scad'.
WARNING: Can't open library 'cascade-C.scad'.
WARNING: Can't open library 'cascade-D.scad'.
WARNING: Can't open include file 'cascade-B.scad'.
WARNING: Can't open library 'cascade-C.scad'.
WARNING: Can't open library 'cascade-D.scad'.
Compiling design (CSG Tree generation)...
WARNING: Ignoring unknown module 'B', in file cascadetest.scad, line 1.
Compiling design (CSG Products generation)...
Geometries in cache: 9
Geometry cache size in bytes: 176616
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized CSG tree has 3 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds

It rendered three objects, B missing.

I don't understand how this is supposed to work. When it detects changes it should keep checking every 200ms to see if anything else changes and then reload and compile. It currently reloads the main file and parses it and then waits for other changes.

Test15: pass apart from the WARNING that I removed.

Test16: pass

Test17: pass with #2821

Test18: pass

Test19: pass. Messages a little different.
```
ERROR: Parser error in file "D:/msys64\home\ChrisP\openscad\testdata\modulecache-tests\error.scad", line 3: syntax error 
Used file cache size: 2 files
Compiling design (CSG Tree generation)...
WARNING: Ignoring unknown module 'errmod', in file mainuseincludingerror.scad, line 2.
TRACE: called by 'errmod', in file mainuseincludingerror.scad, line 2.
```

nophead, 2019-02-27:
As behaviour is generally better with stop on first warnings turned off I think we should also be able to turn off stop on error.

I just can't see how throwing exceptions is not going to break things when all the code was written assuming print was not going to throw.

nophead, 2019-03-25:
I have updated the test procedure and now all the module-cache tests pass. These were mostly fixed by #2843 and #2821.  The README changes are for changes to the messages, etc.

closed 2019-03-28 by kintel