openscad/openscad #2814: Work towards a functioning MSVC Build

thehans, 2019-02-25:
I'm trying to get a build to work natively within Visual Studio 2017 Community Edition.   

I have commited some initial changes here if anyone wants to try for themselves:
https://github.com/thehans/openscad/tree/msvc_build

This is attempting to use [VCPKG](https://github.com/microsoft/vcpkg) for dependencies, since it seem to support nearly all OpenSCAD dependencies, aside from OpenCSG and Lib3MF.

Below are the current steps I've gone through so far, I have not yet successfully built a binary.
I will update/edit this top post with more instructions as new things are figured out.

1) Get Visual Studio 2017 Community Edition (this has some integration with cmake)
https://visualstudio.microsoft.com/downloads/

2) Get Chocolatey for flex/bison tool packages:
https://chocolatey.org/install#installing-chocolatey
choco install winflexbison3

3) Set up VCPKG
```
git clone https://github.com/Microsoft/vcpkg.git
cd vcpkg
.\bootstrap-vcpkg.bat
# Next command integrates VCPKG with VS somehow (will prompt for elevated priviledges)
.\vcpkg integrate install
# Set an environment variable for your vcpkg path.  OpenSCAD's CMakeSettings.json will set CMAKE_TOOLCHAIN_FILE based on this
setx VCPKG_PATH=%cd%
# Will need to restart Visual Studio (if already open) to pick up new environment variable
```
4) Build/Install OpenSCAD dependencies with VCPKG
vcpkg defaults to target `x86-windows`.  If you want to install packages for another target, set environment variable `VCPKG_DEFAULT_TRIPLET` to something else before running the install command below. (run `.\vcpkg help triplet` to see available values)
```
.\vcpkg install boost-assign boost-thread boost-program-options boost-filesystem boost-system boost-regex cgal double-conversion eigen3 fontconfig freetype glew glib harfbuzz libxml2 libzip mpfr qscintilla qt5-base qt5-multimedia qt5-gamepad
```

5) Open OpenSCAD project
git clone https://github.com/thehans/openscad/
cd openscad
git checkout msvc_build
Start VIsual Studio
File -> Open -> CMake...
Select `openscad/CMakeLists.txt`

6) Try building
I have commited a `CMakeSettings.json` file which is used by Ninja which handles the different cmake build configurations.  In this, I've currently enabled `NULLGL` option from the OpenSCAD `CMakeLists.txt`, which is basically the bare minimum functionality that should still be buildable, since OpenCSG has not been packaged yet for VCPKG.
I think VS should pick up on that the existence of the `.json` file, so with any luck you will have 4 Configuration options in the toolbar dropdown, make sure the correct one is selected based on the VCPKG triplet used.  Then try CMake -> Build All

Currently I'm able to get to the linking step, but failing on these:
```
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\openscad.cc.obj : error LNK2019: unresolved external symbol _libintl_textdomain referenced in function _textdomain
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\openscad.cc.obj : error LNK2019: unresolved external symbol _libintl_bindtextdomain referenced in function _bindtextdomain
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\openscad.cc.obj : error LNK2019: unresolved external symbol _libintl_bind_textdomain_codeset referenced in function _bind_textdomain_codeset
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\Camera.cc.obj : error LNK2019: unresolved external symbol _libintl_gettext referenced in function __catch$?viewAll@Camera@@QAEXABV?$AlignedBox@N$02@Eigen@@@Z$2
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\export.cc.obj : error LNK2001: unresolved external symbol _libintl_gettext
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\export_png.cc.obj : error LNK2001: unresolved external symbol _libintl_fprintf
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\handle_dep.cc.obj : error LNK2001: unresolved external symbol _libintl_fprintf
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\printutils.cc.obj : error LNK2001: unresolved external symbol _libintl_fprintf
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\lexer.cxx.obj : error LNK2001: unresolved external symbol _libintl_fprintf
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\parser.cxx.obj : error LNK2001: unresolved external symbol _libintl_fprintf
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\comment_parser.cxx.obj : error LNK2019: unresolved external symbol "void __cdecl comment_lexer_scan_string(char const *)" (?comment_lexer_scan_string@@YAXPBD@Z) referenced in function "class std::shared_ptr<class Expression> __cdecl CommentParser::parser(char const *)" (?parser@CommentParser@@YA?AV?$shared_ptr@VExpression@@@std@@PBD@Z)
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\OpenSCAD.exe : fatal error LNK1120: 6 unresolved externals
  ninja: build stopped: subcommand failed.
```

I think these are provided by `gettext` package, but am unable to get CMake to find the installed package from VCPKG.  
I've opened an issue for that: https://github.com/Microsoft/vcpkg/issues/5435

thehans, 2019-02-25:
Just a note, I've been tinkering around a lot with `CMakeLists.txt` for this, so I know its in a bad state for other platforms right now.  I will attempt some cleanup/refactoring once the Windows build is better nailed down.

thehans, 2019-02-25:
So I was having issue finding `gettext` but realized its AKA `libintl`, so those linker errors are gone.  

The one remaining linker error at the moment is something with the comment_lexer.

Initially, in order to get the build past various compile errors, I added these two options:
 `--wincompat` and `--nounistd` when calling `win_flex.exe`

And also copied this little piece of preprocessor checks from `lexer.l` into `comment_lexer.l`:
https://github.com/thehans/openscad/commit/ea8ed6b28ef835d5456559b5f8d88c0e84fb4d96#diff-230bba01c91025e179018f2ab6f56d90

I guess those three things aren't the exact correct solution to the issue, I'll have to dig more into understanding whats going on here.

thehans, 2019-02-25:
Also I've done some work towards creating a VCPKG port for OpenCSG here: 
https://github.com/thehans/vcpkg/commit/2f86663b3e13f0d2f5961ed26069bd679126ce1c

I had to change a single character from upstream OpenCSG because vcpkg expects UTF-8 and it had an odd high ASCII symbol in the comments of opencsg.h 
So my fork of VCPKG pulls opencsg from my own fork of OpenCSG for now.  
If you check out my fork of VCPKG to try it, you have to run 
`.\vcpkg install opencsg --head`, to make it grab from head without a specific release package/checksum.

I have the build close to working(it compiles), but honestly don't understand a whole lot about the proper way to create a library with cmake.  Attached is the output from vcpkg complaining that my DLLs don't export anything: [vcpkg_install_opencsg.log](https://github.com/openscad/openscad/files/2900373/vcpkg_install_opencsg.log)

kintel, 2019-02-25:
I'm not sure OpenCSG supports being build as a DLL. Try building it as a static library

thehans, 2019-02-26:
OK, made the package static only and it seems to be good now!  I removed `-DNULLGL` from build configs and fixed a few other compile errors that popped up.  Basically just that comment_lexer linking issue is remaining.

thehans, 2019-02-28:
Got the builds working reliably for me now, briefly tested both x86 and x64, Debug and Releases and rendered a few scripts without crashing.  

A few remaining things to do:
- Lib3MF support is still disabled, need to create a vcpkg port for that.
- Its not copying any of the subdirectories into the build output path: "colorschemes", "examples", etc
- No build for console application `openscad.com`
- Still need to get `tests\CMakeLists.txt` running
- Need to test openscad\CMakeLists.txt` on linux, other platforms again to see if I broke anything there.  Could probably still use some cleanup/refactoring now that windows build works.

kintel, 2019-02-28:
We could set up an occasional cmake build for Linux and Mac on Travis. A nightly (or even weekly if we're short on build resources) build should be enough to eventually catch build system bugs

thehans, 2019-02-28:
Well there is PR #2764 that I started working on.  Have you seen it?  Its not quite ready and working yet.  For some reason AMF tests were crashing or something on Trusty, but not Xenial.  More comments about it in the PR, but I kinda got stuck there.

kintel, 2019-02-28:
k, I'll take a look at that when I have some time to burn :)

thehans, 2019-02-28:
Yay, my opencsg port was merged into Microsoft's repo!  Instructions in first post have been updated to clone directly from them now.

Scopeuk, 2019-03-01:
I have replicated this with windows 10 home.
Its worth noting during the visual studio community install you have to select your language tool (c++ in this case) and confirm you have cmake in the list of things that will be installed on the right hand side.
Curriously on loading a model the preview didn't render until i hit f5.

nophead, 2019-03-01:
I have just merged a fix for that into master. #2821

nophead, 2019-03-03:
I also got this to build and run on Win7/64, thanks.

Is it ready for a merge into master?

thehans, 2019-03-05:
@nophead Just about ready I think; requesting comments now for one last thing that was bugging me.  Also I just commited a couple fixes for linux cmake so that it builds and passes all tests again.

There's also still the remaining tasks from my comment above, but I suppose I can track those in a separate issue: https://github.com/openscad/openscad/issues/2814#issuecomment-468119651

nophead, 2019-03-05:
> suppose I can track those in a separate issue: #2814 (comment)

Great. I am in transit ATM but I will try to test the latest on my laptop
tomorrow. I did manage to build it yesterday ok.

Btw msys2 is currently broken by gcc8.3.0 failing to compile some Qt stuff
with an internal compiler error.

thehans, 2019-03-10:
Closing this since the changes are merged now, created new home for further development: #2857

closed 2019-03-10 by thehans