openscad/openscad #2822: Feature request.: Macros for architectural use cases.

afarlie, 2019-02-26:
This is a long term request, but I'd like to see a program in which i was able to specify model buildings or the architectural components  using a set of macros or primitives.

My reason for asking is that I used a program called "Designer Castles" (https://en.wikipedia.org/wiki/Designer_Castles) on a BBC Micro system many years ago and hadn't seen anything like it since on modern platforms.

In this program a "Castle" was built up as a series of castles components such as Towers, Walls, Keeps and certain other items...

As a full blown 'model village' builder is out of scope for this project ( although if someone wants to attempt it,  I'm sure it would gain my interest.)  my request here is limited to a request for a set of primitives, (or possibly better a set of Macros) that could be used to create a 'solid' building, which could then be either 

1. "unfolded" into a set of construction parts in third party software, or onto which other artwork could be texture mapped in other software.

2.  Created directly using 3D printed techniques to make model parts, which could be assembled much in the way you would assemble plastic model kits ( Faller and Pola being some proprietary model kit brands I had in mind).

I appreciate that the required support may not be implemented in the medium term though, and if they is already a 'free' software tool that would better support a 'model village' creator, feel free to recomend it.

t-paul, 2019-02-26:
I'm not sure this is the right place for what you are asking. It seems to be either a program that is totally different from OpenSCAD right now, or about actual usage of OpenSCAD.

For the second part (with focus of 3d printing), see:
* Medieval Fortress Generator: https://www.thingiverse.com/thing:1639567
* Grand Castle Generator: https://www.thingiverse.com/thing:1682427

kintel, 2019-03-09:
Closing as out of scope

closed 2019-03-09 by kintel