openscad/openscad #2825: Module recursion causes crash again.

nophead, 2019-02-28:
module fred() fred();
fred();

Crashes Win7 64 bit, RC2 and more recent.

nophead, 2019-02-28:
Also function recursion:
```
function fred() = fred();
x = fred();
```

nophead, 2019-02-28:
Looks like this line should use std::llabs() for a 64 bit system but that doesn't seem to fix it:
https://github.com/openscad/openscad/blob/master/src/stackcheck.cc#L25

thehans, 2019-02-28:
Hey this Visual Studio build I've been working on already came in handy.
```
 	Qt5Guid.dll!_chkstk() Line 99	Unknown	Non-user code. Symbols loaded.
 	Qt5Guid.dll!QTextLayout::draw(QPainter * p, const QPointF & pos, const QVector<QTextLayout::FormatRange> & selections, const QRectF & clip) Line 1101	C++	Symbols loaded.
 	Qt5Widgetsd.dll!QPlainTextEdit::paintEvent(QPaintEvent * e) Line 2022	C++	Symbols loaded.
 	Qt5Widgetsd.dll!QWidget::event(QEvent * event) Line 9073	C++	Symbols loaded.
 	Qt5Widgetsd.dll!QFrame::event(QEvent * e) Line 550	C++	Symbols loaded.
 	Qt5Widgetsd.dll!QAbstractScrollArea::viewportEvent(QEvent * e) Line 1221	C++	Symbols loaded.
 	Qt5Widgetsd.dll!QAbstractScrollAreaPrivate::viewportEvent(QEvent * event) Line 113	C++	Symbols loaded.
 	Qt5Widgetsd.dll!QAbstractScrollAreaFilter::eventFilter(QObject * o, QEvent * e) Line 129	C++	Symbols loaded.
 	Qt5Cored.dll!QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject * receiver, QEvent * event) Line 1186	C++	Symbols loaded.
 	Qt5Widgetsd.dll!QApplicationPrivate::notify_helper(QObject * receiver, QEvent * e) Line 3745	C++	Symbols loaded.
 	Qt5Widgetsd.dll!QApplication::notify(QObject * receiver, QEvent * e) Line 3704	C++	Symbols loaded.
 	OpenSCAD.exe!OpenSCADApp::notify(QObject * object, QEvent * event) Line 55	C++	Symbols loaded.
 	Qt5Cored.dll!QCoreApplication::notifyInternal2(QObject * receiver, QEvent * event) Line 1060	C++	Symbols loaded.
 	Qt5Cored.dll!QCoreApplication::sendSpontaneousEvent(QObject * receiver, QEvent * event) Line 1462	C++	Symbols loaded.
 	Qt5Widgetsd.dll!QWidgetPrivate::sendPaintEvent(const QRegion & toBePainted) Line 5675	C++	Symbols loaded.
 	Qt5Widgetsd.dll!QWidgetPrivate::drawWidget(QPaintDevice * pdev, const QRegion & rgn, const QPoint & offset, int flags, QPainter * sharedPainter, QWidgetBackingStore * backingStore) Line 5619	C++	Symbols loaded.
 	Qt5Widgetsd.dll!QWidgetBackingStore::doSync() Line 1418	C++	Symbols loaded.
 	Qt5Widgetsd.dll!QWidgetBackingStore::sync() Line 1205	C++	Symbols loaded.
 	Qt5Widgetsd.dll!QWidgetPrivate::syncBackingStore() Line 1957	C++	Symbols loaded.
 	Qt5Widgetsd.dll!QWidget::event(QEvent * event) Line 9237	C++	Symbols loaded.
 	Qt5Widgetsd.dll!QMainWindow::event(QEvent * event) Line 1340	C++	Symbols loaded.
 	OpenSCAD.exe!MainWindow::event(QEvent * event) Line 1803	C++	Symbols loaded.
 	Qt5Widgetsd.dll!QApplicationPrivate::notify_helper(QObject * receiver, QEvent * e) Line 3753	C++	Symbols loaded.
 	Qt5Widgetsd.dll!QApplication::notify(QObject * receiver, QEvent * e) Line 3704	C++	Symbols loaded.
 	OpenSCAD.exe!OpenSCADApp::notify(QObject * object, QEvent * event) Line 55	C++	Symbols loaded.
 	Qt5Cored.dll!QCoreApplication::notifyInternal2(QObject * receiver, QEvent * event) Line 1060	C++	Symbols loaded.
 	Qt5Cored.dll!QCoreApplication::sendEvent(QObject * receiver, QEvent * event) Line 1450	C++	Symbols loaded.
 	Qt5Cored.dll!QCoreApplicationPrivate::sendPostedEvents(QObject * receiver, int event_type, QThreadData * data) Line 1799	C++	Symbols loaded.
 	Qt5Cored.dll!QEventDispatcherWin32::sendPostedEvents() Line 1092	C++	Symbols loaded.
 	qwindowsd.dll!QWindowsGuiEventDispatcher::sendPostedEvents() Line 82	C++	Symbols loaded.
 	Qt5Cored.dll!qt_internal_proc(HWND__ * hwnd, unsigned int message, unsigned int wp, long lp) Line 247	C++	Symbols loaded.
 	[External Code]		Annotated Frame
 	[Frames below may be incorrect and/or missing, no symbols loaded for user32.dll]	Unknown	No symbols loaded.
 	Qt5Cored.dll!QEventDispatcherWin32::processEvents(QFlags<enum QEventLoop::ProcessEventsFlag> flags) Line 640	C++	Symbols loaded.
 	qwindowsd.dll!QWindowsGuiEventDispatcher::processEvents(QFlags<enum QEventLoop::ProcessEventsFlag> flags) Line 74	C++	Symbols loaded.
 	Qt5Cored.dll!QCoreApplication::processEvents(QFlags<enum QEventLoop::ProcessEventsFlag> flags) Line 1280	C++	Symbols loaded.
 	OpenSCAD.exe!MainWindow::processEvents() Line 3159	C++	Symbols loaded.
 	OpenSCAD.exe!MainWindow::consoleOutput(const QString & msg) Line 3131	C++	Symbols loaded.
 	OpenSCAD.exe!MainWindow::qt_static_metacall(QObject * _o, QMetaObject::Call _c, int _id, void * * _a) Line 555	C++	Symbols loaded.
 	Qt5Cored.dll!QMetaMethod::invoke(QObject * object, Qt::ConnectionType connectionType, QGenericReturnArgument returnValue, QGenericArgument val0, QGenericArgument val1, QGenericArgument val2, QGenericArgument val3, QGenericArgument val4, QGenericArgument val5, QGenericArgument val6, QGenericArgument val7, QGenericArgument val8, QGenericArgument val9) Line 2295	C++	Symbols loaded.
 	Qt5Cored.dll!QMetaObject::invokeMethod(QObject * obj, const char * member, Qt::ConnectionType type, QGenericReturnArgument ret, QGenericArgument val0, QGenericArgument val1, QGenericArgument val2, QGenericArgument val3, QGenericArgument val4, QGenericArgument val5, QGenericArgument val6, QGenericArgument val7, QGenericArgument val8, QGenericArgument val9) Line 1508	C++	Symbols loaded.
 	Qt5Cored.dll!QMetaObject::invokeMethod(QObject * obj, const char * member, QGenericArgument val0, QGenericArgument val1, QGenericArgument val2, QGenericArgument val3, QGenericArgument val4, QGenericArgument val5, QGenericArgument val6, QGenericArgument val7, QGenericArgument val8, QGenericArgument val9) Line 460	C++	Symbols loaded.
 	OpenSCAD.exe!MainWindow::consoleOutput(const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & msg, void * userdata) Line 3100	C++	Symbols loaded.
[... cut ...]
https://github.com/openscad/openscad/blob/40c257529d31f92e2bea2f7b1ded3c1f5da64c68/src/UserModule.cc#L45-L51

The print_trace call here still triggers it:
https://github.com/openscad/openscad/blob/40c257529d31f92e2bea2f7b1ded3c1f5da64c68/src/ModuleInstantiation.cc#L93-L100

A quick fix might be to just increase our "stack buffer size".  But a more complete fix would probably be specially handle RecursionException in a separate catch that keeps throwing until we're safely down the stack again before attempting to print any errors.

nophead, 2019-02-28:
Doubling STACK_BUFFER_SIZE to 128K seems to fix it. Unbelievable that printing via QT takes that much stack.

nophead, 2019-02-28:
There are regression tests but I don't think the MXE builds run tests do they?

t-paul, 2019-02-28:
No, only AppVeyor runs the test suite (with some exclusions). This is MSYS2 based.

nophead, 2019-02-28:
It crashes for me in MSYS2 but of course the command line version doesn't use QT, so it won't use anything like as muck stack to print.

The regression tests check that the recursion is detected and prints the correct information but it doesn't validate the STACK_BUFFER_SIZE value. I think that has to be trial and error in the GUI. 

Doubling it until it stops crashing seems reasonable because 128K is still small compared to the 8MB stack.

nophead, 2019-02-28:
It is a general problem that the GUI code has no automatic test regime. It is currently quite broken after some big feature changes. Unfortunate when we are supposed to be close to a release.

I don't really have any experience of GUI testing. The last big GUI application I wrote in MFC was tested by building a Python interpreter into it that could call all the C++ methods via boot wrappers. We could connect to the app and control it with python scripts. It wasn't quite the same as interacting with the GUI with mouse and keyboard, we were invoking the functions that would have been called had we done that. The GUI did respond though and update.

Surely they are  better ways of auto testing GUIs these days, nearly 15 years later.

nophead, 2019-02-28:
Fixed via #2828

closed 2019-02-28 by nophead

thehans, 2019-02-28:
Yeah I wasn't thinking about the fact that the automated tests are all command line, duh!  I wonder if it would even be possible to have a subset of tests that runs gui on headless server but renders the window to a framebuffer or something, perhaps with a special command line option.

MichaelAtOz, 2019-02-28:
There is the likes of [Autohotkey](https://www.autohotkey.com/), for Windows. Unfortunately not multi-platform.
It is not specifically designed as a testbed, but is full screen-scraper GUI driver capable (and much more).
I used to be fluent, it's... 'non-standard' may be the best description of the language.

thehans, 2019-02-28:
Even some basic tests that automatically close the window after exporting(not requiring any user input/interface scripting hopefully) could be useful.