openscad/openscad #2826: Stop on first error doesn't play well with auto reload

nophead, 2019-02-28:
The dependency checking code in MainWin is quite fragile and assumes it runs to completion. When it gets interrupted by an exception that prints Execution Aborted the next auto update carries on where it left off and does some more and will eventually try to render the result, exactly what stop on error is supposed to prevent. So it looks to the user like execution is far from aborted.

This was masked by the fact that the auto reload timer didn't get restarted but since that was fixed we now have this opposite problem. 

When it is checking dependencies it normally checks them all and updates the mtime so they don't get checked again unless they have been changed. The exception will prevent the mtime being updated so it starts again 200ms later.

nophead, 2019-02-28:
The parse of the main fail and the handleDependencies stuff needs to run to completion. So here is my proposed solution:

Add a defer_exceptions() call to printutils that stops PRINT throwing but counts errors and warnings. This will be called in compile before parsing the main file.

Add a second function throw_deferred_exceptions() that checks the counters and throws if there would have been any. Call that after the parsing and dependency checks but before it does instantiation and rendering.

I think that will fix the auto reload problems. It will mean that stop on first warning will actually stop after all parser warnings but it is quite fast. I could suppress the messages but that would hide some errors and be misleading.

It should fix the problems of it not actually stopping at all.

If this is acceptable and I can make a PR but I have limited time as I go on holiday again next week.

nophead, 2019-03-01:
@MichaelPFrey, I was expecting PRINT to throw when an error is printed but I only see code to throw on warnings. Parser errors don't seem to attempt to stop execution whereas warnings do.

How is stop on error handled?

MichaelPFrey, 2019-03-02:
> @MichaelPFrey, I was expecting PRINT to throw when an error is printed but I only see code to throw on warnings. Parser errors don't seem to attempt to stop execution whereas warnings do.
> 
> How is stop on error handled?
Currently inconsistently.

Some errors like what we currently call recursion (which is in reality hitting an 
arbitrary limit or stack exhaust) throw their own exception.
Parser errors stop the code execution by stopping the parsing.

Other errors simply continue.

We could throw an hardwarning exception up on encountering "error". Doing that would require some thinking about if the exceptions could/would interfere and some practical testing. (which could be tricky, as ideally a warning is triggered because of bad parameter, preventing the error from occurring later)

nophead, 2019-03-03:
Thanks for the clarification. I am currently setting a flag in printutils that prevents parser warnings throwing but instead suppresses subsequent messages to fake that it has stopped but actually carries on and then stops further processing after parsing is finished. Errors do the same, but only when the flag is set. So basically during parse it looks from the outside that warnings and errors stop.

It is mainly working but I have a bug to chase down where it shows a syntax error twice in and include in a use.

>Other errors simply continue.

Perhaps they should be changed to say WARNING: Otherwise it is odd if you have stop on warnings enabled and some errors don't stop.

nophead, 2019-03-24:
Fixed by https://github.com/openscad/openscad/pull/2843

closed 2019-03-24 by nophead