openscad/openscad #2827: Simple Editor -> "save" to reload after every Customizer change

paulirotta, 2019-02-28:
[image]

The Customizer is greyed out in Windows 10 after successfully updating and making a change. It re-appears after CTRL S save.

This happens in the Simple Editor but not QScintilla Editor. This happens even in the Simple Editor view is closed (having Simple Editor set in Preferences is sufficient to surface the problem).

MichaelAtOz, 2019-02-28:
> successfully updating and making a change

Does this mean typing then F5?  Or changing Customizer value?

MichaelAtOz, 2019-02-28:
Confirmed on W7/64.
Grey's out on changing item in Customizer, image is refreshed, stays grey.
Ditto typing in editor.
Note  grey's out when typing in the QScintilla too, but I presume this is intention, as it goes away after F5.

paulirotta, 2019-03-04:
F5 is not needed to trigger it- just a Customizer value

t-paul, 2019-11-11:
Closing as the simple editor will not be included in the next release anymore.

closed 2019-11-11 by t-paul