openscad/openscad #2829: GUI testbed

MichaelAtOz, 2019-03-01:
There is the likes of [Autohotkey](https://www.autohotkey.com/), for Windows. Unfortunately not multi-platform.
It is not specifically designed as a testbed, but is full screen-scraper GUI driver capable (and much more).
I used to be fluent, it's... 'non-standard' may be the best description of the language.

_Originally posted by @MichaelAtOz in https://github.com/openscad/openscad/issues/2825#issuecomment-468472862_

MichaelAtOz, 2019-03-01:
@thehans 
> Even some basic tests

List a few, I'll give it a go.

MichaelAtOz, 2019-03-01:
...all though, I'll have to double check, I think QT way not list the Window components well. But maybe a MSVC build will be friendlier.

thehans, 2019-03-01:
By basic tests, I meant those such as the test from the original discussion(It looks like the we have these under `recursion-test-module.scad` and `recursion-test-function2.scad`), where we just want to know if it crashed or not, and aren't necessarily testing clicking through GUI menus/interaction etc.

So this sort of test would ideally not require any GUI scripting, if we can have some command line parameter that forces GUI creation (call it `-force-gui` or some such) when you are running command line that would normally just export without creating a window.  Also this implies that implementation of such a flag would still automatically exit the program after export was complete.

Interface scripting may still be useful for actually verifying that particular options/menus etc work as expected but this was more an idea to get something up and running faster without integrating a whole new third party tool, maybe as a stepping stone towards that tool though.

MichaelAtOz, 2019-03-01:
It would be easy to load a .scad file, do F5 &&/|| F6 (or any menu or keystrokes), monitor the window to see if the below happens
[image]
If so capture the crash details and click 'close the program' button.
If not grab the console output, either store it or capture some specific bits and save them too.

I'm going to do a simple prototype, because one of the things I'm concerned about is performance changes, ie worse CPU or memory creeping into changes. We have nothing for that ATM.

thehans, 2019-03-01:
Yeah I forgot about crash dialogs, maybe it is a necessity to have such a scripting engine just to close that dialog.

t-paul, 2019-03-01:
Looks like the Accessibility features of Qt5 are now included by default and working, including in QScintilla. I've tested on Linux, but assume this would be also available on Windows. As we have much easier access to Linux based hosted servers I'll probably give it another try using that for GUI tests. Last time that got stuck as there were pretty big show stoppers like the editor not being visible at all.

For Windows we currently only have AppVeyor and that is pretty slow as it does not allow any concurrently running jobs - maybe there would be the option to run the tests via a different repo and account just once a day or so. I do believe I have seen AutoHotkey installed there.

kintel, 2019-03-02:
For Qt, I've hear that the "best in class tool is Squish: https://www.froglogic.com/squish/editions/qt-gui-test-automation/

MichaelAtOz, 2019-03-02:
> Example 2
>
>   Group of 5 individuals creating, editing or debugging tests
    Up to 2 group members expected to use Squish concurrently.
    Squish needed for 2 Squish editions (i.e., Windows, Mac).
    Unlimited automated, hands-off test execution included
Up-front license fee for this configuration is 2139 € per user (total of 10695 €). This includes a full 12 months of support & updates subscription.

kintel, 2019-03-02:
Yah, but last I checked I believe they were free for Open Source projects.

MichaelAtOz, 2019-03-02:
> Looks like the Accessibility features of Qt5 are now included by default and working,

Not on Windows, this is what is 'visible' 
[image]

With accessible apps there is much text and each GUI component (icon/menu etc) is addressable.
It may be build settings, but ATM Qt Widget seems to be the finest granularity, ie there is a ClassNN for each panel, but everything inside is not build by Windows GUI components.

MichaelAtOz, 2019-03-07:
Seems there is a separate accessible mechanism which sort of works with Windows. 
Running  Accessories/Ease-of-access/Narrator speaks some things, but much is missing. Try it with Preferences for example.

I now have an advanced prototype GUI driver.

It can:

detect a crash 
[image]

do F5, F6 and export 
[image]
(note at 1. export/image was done)

That was driven by
```
  tdPreview()
  tdExport("image")
  tdRender()
  tdExport("stl")
  tdExport("OFF") ; stl,off,amf,3mf,dxf,svg,csg,image
;
  tdSendKeys("^q","")
```

I just need to parse the output & decide how best to drive it to do various tests.
I'm also looking at sysinternals pslist.exe to get the CPU & Memory usage, looks reasonably easy.