openscad/openscad #2830: Feature request - command line option --nowelcome, or --welcome=off

MichaelAtOz, 2019-03-01:
To facilitate testing via the GUI it would be nice to be able to run a specific instance without the Welcome Screen, without having to disable the setting for normal interactive use.
Yes this could be achieved via `openscad somefile.scad` but that would not allow for testing File/Open (or anything else) from a empty clean start state.

MichaelAtOz, 2019-03-01:
A more complex solution to allow any preference to be specified as a temporary override could be considered too. So maybe `--preference=<preference>,<value>`??

t-paul, 2019-03-01:
I see that this might make things easier for testing, but it also adds complexity (especially considering Commmand-Line and GUI don't share settings at all currently) and will cause code path to be tested that the normal user will not actually use. There are probably ways to reach the same goal (be able to define a specific test environment) without adding lots of special code.

MichaelAtOz, 2019-03-07:
ATM I'm doing `openscad blank.scad' to bypass the welcome screen.

t-paul, 2019-03-07:
I wonder if #670 would be a good idea and help in this case to plug in defined config files for specific tests.