openscad/openscad #2833: Format warnings - delimit variable names etc.

MichaelAtOz, 2019-03-02:
From [forum](http://forum.openscad.org/New-Release-of-OpenSCAD-soon-please-test-tp25648p25721.html). I tend to agree. 
The file name should also be delimited as it could be called something confusing too, perhaps just a `:<space> ` before it assuming it is always at EOL.

> Regarding syntax checking, the following code outputs an odd waring

```
function somefunc(size) = 2 * size;
a = somefunc(size = 1, type = "foo");
```

> $ openscad-nightly -o out.stl bug-1.scad
> WARNING: variable type not specified as parameter, in file bug-1.scad, 
> line 3 Current top level object is empty.

> Formatting could perhaps be improved to more easily tell variable names from the actual warning text.
> Like wrapping the variable name in quotes, angle brackets, something like that.

MichaelAtOz, 2019-03-05:
While I'm at it, and I know this is nit picking, but
`UI-ERROR: Current top level object is not a 2D object.`, `UI-ERROR:` seems a bit cludgy.

```

	if (msg.startsWith("WARNING:") || msg.startsWith("DEPRECATED:")) {
		this->compileWarnings++;
		this->console->appendHtml("<span style=\"color: black; background-color: #ffffb0;\">" + QT_HTML_ESCAPE(QString(msg)) + "</span>");
	} else if (msg.startsWith("UI-WARNING:") || msg.startsWith("FONT-WARNING:") || msg.startsWith("EXPORT-WARNING:")) {
		this->console->appendHtml("<span style=\"color: black; background-color: #ffffb0;\">" + QT_HTML_ESCAPE(QString(msg)) + "</span>");
	} else if (msg.startsWith("ERROR:")) {
		this->compileErrors++;
		this->console->appendHtml("<span style=\"color: black; background-color: #ffb0b0;\">" + QT_HTML_ESCAPE(QString(msg)) + "</span>");
	} else if (msg.startsWith("EXPORT-ERROR:") || msg.startsWith("UI-ERROR:") || msg.startsWith("PARSER-ERROR:")) {
		this->console->appendHtml("<span style=\"color: black; background-color: #ffb0b0;\">" + QT_HTML_ESCAPE(QString(msg)) + "</span>");
	} else if (msg.startsWith("TRACE:")) {
		this->console->appendHtml("<span style=\"color: black; background-color: #d0d0ff;\">" + QT_HTML_ESCAPE(QString(msg)) + "</span>");
} else {
```
I see the prefix is used internally.
What if the prefix is stripped off before sending to the console?
Then you just get `WARNING:` OR `ERROR:`, looks cleaner IMO.

MichaelPFrey, 2019-03-05:
>  `UI-ERROR:` seems a bit cludgy.

Yes, that is a cludge. Hardwarning trows an exception up on seeing "WARNING". My plan was kind of to do the same thing with errors, (some errors currently stop further execution, some do not) but I am currently to concerned that when doing that I miss possible execution paths, leading to an uncaught exception.

closed 2025-01-05 by kintel

jordanbrown0, 2025-01-05:
PR#5562 addresses the top-line complaint here, ambiguity around variable names.

New issue #5566 covers file names.
New issue #5567 covers module and function names.
New issue #5568 covers UI-ERROR.

commit bf9188fb (2025-01-05): Consistent quoting of variable names (via central function).  Fixes #2833. (#5562)

* Consistent quoting of variable names (via central function).  Fixes #2833.
* Add TODO note for duplicate-assignment messages.