openscad/openscad #2841: Minkowski rendering error.

revarbat, 2019-03-04:
I just ran into a really bizarre bug, apparently in `minkowski()`.  The following code renders incorrectly on my OpenSCAD version 2019.03.03 under OSX, in both F5 preview and F6 rendering:

    $fn = 7;
    minkowski() {
        cube(0.1);
        union() {
            translate([5, 0, 0]) cylinder(h=10, r=5, center=true);
            cylinder(h=5, r=10, center=true);
        }
    }

It should output a slightly warped wide cylinder with a longer, thin cylinder poking through it along one side. Instead it shows what looks like the longer, thin cylinder, with the wider cylinder differenced from it.  It acts like that union() is a difference().

It'll render correctly if you remove the translate() or if you translate by a value other than 5.  It works if you set $fn to any of 5, 6, 8, 9, or 12, but it fails when $fn is 7, 10, or 11.

Here's an image of the failed render:
[image]

Here's an image of a successful render with $fn=8:
[image]

thehans, 2019-03-05:
Well, the good news is at least it doesn't crash.  I tested if this was a regression and release 2015.03 just crashes on this input.

henwist, 2019-06-14:
Tested on OpenSCAD 2019.05 run from the console and I have got an error to stdout:

> CGAL::Polyhedron_incremental_builder_3<HDS>::
lookup_halfedge(): input error: facet 2 has a self intersection at vertex 7.

from this code in editor:

> $fn = 7;
minkowski() {
    cube(.1, center=true);
    

    union() {
            translate([5, 0, 0]) cylinder(h=10, r=5, center=true);
        cylinder(h=5, r=10, center=true);
    }
}

So this bug seems to do with self intersections when minkowski is in use - the union is rendering just fine alone. When I translate the first cylinder in x to e.g. 5.0002, then the rendering is correct again.

nophead, 2019-06-14:
The problem seems to be unioning the two heptagons with nearly coincident faces, but not quite, because sin and cos of 360/7 are irrational numbers. If you turn on show edges you can the joins.

I wouldn't call it a bug because that will never work. Faces and vertices need to be exactly coincident or significantly different. If they are only different in the floating point rounding of irrational numbers CSG will fail.

dstulken, 2019-06-18:
Just recently I've hit this issue myself, with model features silently disappearing when using minkowski().

Here is my test case which reproduces the bug.  

```
minkowski()
{
	union()
	{
		translate([-15, -15, 2])
		{
			cube([30, 30, 4]);
		}

		difference()
		{
			cylinder(8, 10, 10, false, $fn=8);
			translate([0,0,-1])
			cylinder(10, 9, 9, false, $fn=8);
		}
	}

	sphere(0.5, $fn=8);
}
```

This code should render as a hollow tube bisected by a center-aligned cube (flat square shape), all edges of all shapes being slightly rounded.
Instead, the flat square disappears completely, and the remaining two parts of the tube have their edges rounded.  

Similar to @revarbat's test case, this bug makes union() act like a difference(). This is clearly visible in my test case, as both sides of both rings formed by the removal of the cube have been rounded by the minkowski() operation.  So the minkowski() operation is being performed AFTER the geometry removal.   Curiously though, my code has the cube first and the cylinders second, so the union() isn't *quite* becoming a difference() - the components are in the wrong order.

@nophead, I believe my case should be free of coincident faces.  The cylinder(s) extend beyond the cube on both sides, and likewise the cube is of distinctly larger x/y dimensions than the cylinders that make up the tube.   

This renders incorrectly on both OpenSCAD 2019.03.03 and 2019.05, using the Mac builds.  Neither version crashes, and neither version shows any warnings in the console window.  

In my test case, I don't see any differences when changing the value of $fn.  Only changing the translate parameters to move the cube (+10 in the X dimension for example) causes the rendering to succeed.

nophead, 2019-06-18:
I agree that looks like a bug. 

I think Minkowski of concave shapes involves convex decomposition, doing the convex parts separately and then unioning the result. Looks like the cube gets lost in that process somehow.

If you need a workaround manually doing the Minkowski on the tube and cube separately and unioning the  results gives the correct shape.

MichaelAtOz, 2019-06-18:
This last example also crashed 2015.03-2.

Exception Code:	40000015
Some digging shows.

> They can result if you try to make a virtual function call from a constructor or destructor. Since you can't make a virtual function call from a constructor or destructor (the derived class object hasn't been constructed or has already been destroyed), it calls the base class version, which in the case of a pure virtual function, doesn't exist.

May be a red-herring going down the old version rabbit hole tho.

dstulken, 2019-06-19:
> May be a red-herring going down the old version rabbit hole tho.

Interestingly enough, I just downloaded OpenSCAD 2011.04 (Mac), and it renders my test case correctly!   So this is indeed a regression.

henwist, 2019-06-20:
I have got an error to stdout on the latest version to date when running dstulken's code:

> CGAL::Polyhedron_incremental_builder_3<HDS>::
add_vertex_to_facet(): vertex index 81 is out-of-range [0,27].

and the resulting object is two rings that are somewhat chamfered.

dstulken, 2019-06-20:
I'm not sure that it matters, but I can also say that while cylinders have been used for both examples, they appear to not be the only geometry which triggers this bug.

Replacing the cylinders with cubes:
```
translate([-5,-5,0])
cube([10, 10, 8]);
translate([-4.5,-4.5,-1])
cube([9,9,10]);
```

... or spheres:
```
translate([0,0,6])
sphere(5);
translate([0,0,9.5])
sphere(5);
```

will also demonstrate the issue.

tturpin, 2020-11-07:
Hi,

I'm also experiencing issues with minkowski. For me it causes an assertion failure in CGAL, not in the minkowski operation itself, but later when doing boolean operations with the result. The example is large, and cannot be easily reduced.

But it has one thing in common with the current issue: it is solved by changing cgalutils-applyops.cc to use Nef minkowski only (the handler code). So I'm wondering: what is the issue with Nef minkowski which makes i worse that the default solution, which clearly doesn't work even in quite simple cases like this one ?

kintel, 2024-03-30:
Status as of 2024-03-24:
* First example works with Manifold, but still fails with CGAL
* Second example fails with both Manifold and CGAL
*

kintel, 2024-07-25:
Reported as a CGAL bug: https://github.com/CGAL/cgal/issues/8379

kintel, 2025-03-24:
Status as of 2025-03-23:
* First example works with Manifold, but still fails with CGAL
* Second example works with both Manifold and CGAL