openscad/openscad #2842: Inconsistent F6 behavior: polyhedra with holed faces

RonaldoCMP, 2019-03-04:
A good way to represent polygons with holes is by what we call a "fake simple polygon" (FSP) (I guess nophead call it a "keyhole representation"). It is a polygon whose edges do not cross but some edges are used twice when the polygon edges are circulated. Her is an example:

p = [ [100,0], [0,100], [-100,0], [0,-100], [100,0], [50,0], [0,-50], [-50,0], [0,50], [50,0] ];

Although a FSP is not a simple polygon, it is acceptable by the primitive polygon() and can be extrude by linear_extrude and rotate_extrude to model a solid with a internal hole.

However, the render of polyhedrons where some faces are FSP are subject to a variety of results under F6. A striking example follows:

r = 0.0001;
verts =  [[0, 0, -50], [0, 50, 0], [0, 0, 50], [0, -50, 0], 
          [100, 0, -50], [100, 50, 0], [100, 0, 50], [100, -50, 0], 
          [0, 0, -12], [0, -12, 0], [0, 0, 12], [0, 12, 0], 
          [200, 50, -50], [177.639, 94.7214, 0], [200, 50, 50], [222.361, 5.27864, 0],
          [200, 50, -12], [205.367, 39.2669, 0], [200, 50, 12], [194.633, 60.7331, 0], 
          [100, -12, 0], [100, 0, 12+r], [100, 12, 0], [100, 0, -12+r]] ;
         
 faces = [ [0, 4, 5, 1],    [1, 5, 6, 2],         // 4 external faces
           [2, 6, 7, 3],    [3, 7, 4, 0], 
           [9, 20, 21, 10], [10, 21, 22, 11],     // 4 internal faces
           [11, 22, 23, 8], [8, 23, 20, 9],      
           [0, 1, 2, 3, 0, 8, 9, 10, 11, 8],      // back face with hole
           [7, 6, 5, 4, 7, 20, 23, 22, 21, 20]];  // front face with hole
 
polyhedron(verts, faces);
cube(50);

As is, this code generates the expected result on F6: a cube added to a extrusion of a polygon with a hole. But if the value of the variable r is changed to zero, the render will show the polyhedron missing the holed faces.

thehans, 2019-03-05:
Your 2d example seems like an odd hack, as you can already represent holes if you supply multiple paths.
```openscad
p = [ [100,0], [0,100], [-100,0], [0,-100], [50,0], [0,-50], [-50,0], [0,50] ];
polygon(points=p, paths=[ [0,1,2,3], [4,5,6,7] ]);
```
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_the_2D_Subsystem#polygon

Anyways your second example renders fine for me with `r=0`.  Tested on 2019.02.20 (git c9e4b125) Windows build.

What version are you using?

RonaldoCMP, 2019-03-05:
>
> Your 2d example seems like an odd hack, as you can already represent holes
> if you supply multiple paths.
>
> p = [ [100,0], [0,100], [-100,0], [0,-100], [50,0], [0,-50], [-50,0], [0,50] ];
> polygon(points=p, paths=[ [0,1,2,3], [4,5,6,7] ]);
>
>
> https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_the_2D_Subsystem#polygon
>
You are right: we always can do a partition of a polygon with holes where
all parts are simple polygon. A radical solution is to triangulate the
polygon. It is easy to define a partition in simple cases like that, with
only one convex hole of convex polygon. For more general cases, we need a
function to do it. I have developed a method in OpenSCAD to find such a
partition. Although it is better than ear-cut triangulation it is rather
intricate. I found also that the OpenSCAD primitive polygon() accepts the
keyhole representation of polygons with holes like the one you considered
an odd hack. I have also developed a method to find a keyhole
representation like this for any polygon, convex or not, with any number of
holes, convex or not. The method is very light and more efficient than
finding a partition in simple polygons. All that was discussed in a thread
of forum.openscad. You may want to look at it from
http://forum.openscad.org/Polyhedron-tube-with-irregular-sides-is-it-possible-tp13813p25151.html
.

> Anyways your second example renders fine for me with r=0. Tested on
> 2019.02.20 (git c9e4b12
> <https://github.com/openscad/openscad/commit/c9e4b125a487bf0dd1f8f55be9de6028ed83770d>)
> Windows build.
>
> What version are you using?
>
I had tested it with version 2019.01-RC2 (git 35858ead) running under
Windows 7/64 bits. I checked it now with version 2019.01.10.ci1115 (git
95f53797) and it rendered fine even with r=0.

I guess the keyhole representation is not palatable to CGAL and a
polyhedron with a keyhole polygon as a face renders fine just when OpenSCAD
triangulate the polygon before sending it to CGAL. But not always
polyhedron faces are previously triangulated which leads to inconsistent
results. The inconsistency would vanish if OpenSCAD always triangulate
polyhedron faces before calling CGAL.

thehans, 2019-03-05:
OK, I tested some more and I can reproduce now.  It seems that if F5 is used before F6, it will work OK.  
If you have used F6 first, then you must flush caches and then F5/F6 will work again.

Preview requires converting Nef Polyhedron to Polyset which requires triangulation.  The slightly odd behavior is due to an optimization I made so that it does triangulation before inserting into cache, whereas before it would do triangulation after pulling from cache, meaning.  
https://github.com/openscad/openscad/blob/266fefa2748bceb304f4aa7189a013f1f81e8caa/src/GeometryEvaluator.cc#L62-L89

If you try 2015.03 for example, you can see how F6 fails no matter what because it never gets triangulated before going into cache.

I suppose we could force it to always triangulate, regardless if being previewed first.

nophead, 2019-03-05:
Currently when you do F6 from a clean cache you get better STLs because it
isn't converted to doubles and back. Please don't do this.

kintel, 2019-03-07:
To add to @nophead's concern: We try to do a clean evaluation in F6 mode to avoid moving any intermediate results out of CGAL's representation (causing us to lose precision and possibly topology). 

I believe the behavior where we get different result depending on whether preview was done before render is a bug: We don't cache different geometry objects for the two use-cases, and end up choosing a rounded geometry even though we're doing a pure CGAL render.

I thought we already had an issue for this, but I cannot find it atm.

kintel, 2019-03-07:
In terms of CGAL: I guess CGAL regards this as a self-intersecting non-manifold, and hence refuses to convert it to a Polyhedron_3. We have a number of similar issues with imported STL files.

RonaldoCMP, 2019-03-07:
Hans L <notifications@github.com> wrote:

> OK, I tested some more and I can reproduce now. It seems that if F5 is
> used before F6, it will work OK.
> If you have used F6 first, then you must flush caches and then F5/F6 will
> work again.
>
It doesn't work for me with version RC2 when the perturbation r is equal or
less than 1e-16.

RonaldoCMP, 2019-03-07:
Marius Kintel <notifications@github.com> wrote:

> In terms of CGAL: I guess CGAL regards this as a self-intersecting
> non-manifold, and hence refuses to convert it to a Polyhedron_3. We have a
> number of similar issues with imported STL files.
>
I understand that but there is a self-intersection in the following change
in the vertex list  and it renders fine for r greater than 9e-15.

verts = [[0, 0, -50], [0, 50, 0], [0, 0, 50], [0, -50, 0],

         [100, 0, -50], [100, 50, 0], [100, 0, 50], [100, -50, 0],

         [0, 0, -12], [0, -12, 0], [0, 0, 12], [0, 12, 0],

         [200, 50, -50], [177.639, 94.7214, 0], [200, 50, 50], [222.361,
5.27864, 0],

         [200, 50, -12], [205.367, 39.2669, 0], [200, 50, 12], [194.633,
60.7331, 0],

         [100, -12+r, 0], [100, 0, 12], [100, 12, 0], [100, 0, -12+r]] ;

Interestingly, CGAL seems to eliminate the self-intersecting edge from the
mesh.

[image: renderEdgesOn.PNG]

I guess that the non-planarity of the self-intersecting face with r>e-16
triggered a face triangulation undoing the self-intersection and CGAL
eliminated all internal edges of the triangulation in the final render. So,
CGAL is able to accept polyhedron faces with holes when they are input in a
proper way.

nophead, 2019-03-07:
If it gets converted to a Polyset that is just a polgon soup. When it is triangulated you probably get a triangle each side of the cut line and that is fine in the soup representation because two triangles sharing a line is how it should be. The cut gets auto healed.

The problem I see is we don't want to force triangulation of planar polygons because CGAL processes them as polygons. When you triangulate in double representation it often makes a planar polygon non-planar due to imprecision and then the end result ends up with more facets and vertices than it needs.

RonaldoCMP, 2019-03-07:
> The problem I see is we don't want to force triangulation of planar
> polygons because CGAL processes them as polygons. When you triangulate in
> double representation it often makes a planar polygon [...]
If it doesn't degrade the process time, what is the trouble to end up with
more facets and vertices than it needs?  To generate STL, all faces will be
triangulated anyway.

thehans, 2019-03-07:
I think boolean operations are something like O(N^3) complexity so CGAL should be given as few facets and vertices as possible for performance reasons. 

It sounds like what we need is to handle the Polyset and Nef caches independently.  Either by storing a pair of Geometries in GeometryCache (and setting/requesting one specific type at a time).  Or just keeping two different cache instantiations.

As a side note, when I moved the tessellation step to go before cache insert, I didn't do this for 2D, because the tesselation code converts from Polygon2D into 2D Polysets, and the code that pulls from cache never expects to get a 2D Polysets back.  Reference: https://github.com/openscad/openscad/issues/2737#issuecomment-458372940
So having 2D Polysets also handled separately from Polygon2D would be nice as well.

None of this addresses the original issue of "keyhole" faces for polyhedron, but I'm not sure if there's a good solution to that.  Could CGAL evaluate faces in some way similar to what tessellation is doing, to determine holes without converting to imprecise representation?

thehans, 2019-03-07:
Actually, now that I think about it, I'm not sure why/how CGAL operations are pulling from GeometryCache when we have already have a separate CGALCache,  I am not at my dev computer right to browse source in my preferred IDE but I can look into it later.

nophead, 2019-03-09:
> If it doesn't degrade the process time, what is the trouble to end up with
> more facets and vertices than it needs?  To generate STL, all faces will be
> triangulated anyway.

You end up with more triangles and edges than needed and flat planes are no longer flat due to rounding. Basically the STLs become messy and do not reflect the simple geometry I have programmed. Also it leads to degenerate triangles and self intersections because of the snapping to doubles and then to a grid and then to floats.

To get clean STLS I either clear the cache before F6 or use the command line.

RonaldoCMP, 2019-03-09:
> You end up with more triangles and edges than needed and flat planes are
> no longer flat due to rounding. Basically the STLs become messy and do not
> reflect the simple geometry I have programmed. [...]
I can't say I understand all your pleas against it but from my experience,
and my last image is an example, CGAL removes unnecessary edges from planar
faces and eventually changes the triangulation of a polyhedron face I had
triangulated.

nophead, 2019-03-09:
CGAL is happy to process polygonal faces, keeping them planar due to its use of rationals. The final result gets triangulated for the STL and has minimum triangles and edges and planar faces are as planar as possible in floats.

When you use F5 the Nef polyhedrons get converted into PolySet polygon soup and the rationals get converted to floats, then converted to doubles, and then snapped to an arbitrary grid. At best planar polygons get made  non planar, at worse topology is broken by self intersections and degenerate triangles and then F6 will fail. See https://github.com/openscad/openscad/pull/2694 for an example. I fixed degenerate triangles but self intersections are more complex.

CGAL can remove edges from perfectly planar faces that are parallel to one of the axes. However if a faces is slanted in floating point representation they are no longer planar to CGAL.

I think if you have a sloping faces in CGAL with something meeting it with intersection or union the points where they meet can be exactly on the plane due to its use of rationals. If you convert to float the intersections points cant' be exactly on the plane so the resulting non-planar polygon has to be triangulated before it can be fed back to CGAL.

kintel, 2024-03-30:
Status as of 2024-03-24:
* CGAL struggles rendering it, but it works with Manifold
* Preview is missing the hole if Manifold was used to render previously.
*

kintel, 2025-03-24:
Status as of 2025-03-25:
* CGAL struggles rendering it, but it works with Manifold