openscad/openscad #2845: Changes to assist GUI scraping,  Console right-click menu, and activity indicator

MichaelAtOz, 2019-03-05:
Note: While this is focused on my GUI scraping ATM, it is equally applicable to Accessibility.

I've got much of the GUI driver working; starting OpenSCAD, detecting exit-code & crashes (saving the reason & closing it), driving the interface for F5/F6, getting console results etc.
The only difficulty, with Qt not using accessible controls, is determining what it is doing.
ATM the only way is iteratively scraping the console to Select-All & Copy, and then checking the Clipboard, until the magic "\n...finished." is detected. Not ideal.

ATM this is via sending Click to the console GUI-control & then Control-A/Control-C.

2015 has the Control-C & A shortcuts 
[image]

2019 appears to have lost them
[image]
and gained two items (without shortcuts AFAICT).

Note for 2019, if the console is clicked, then ^A & ^C does do Select-all and Copy, but that may be happen-stance; I'd like to ensure this is 'supported' into the future.

It would also help to add shortcuts for the two new menu items. I'm not using them ATM, but I suspect Clear will be useful, and Save maybe useful if we need to leave the Clipboard alone and can't get proper accessibility working.

Note [key wiki](https://github.com/openscad/openscad/wiki/Keyboard-Shortcuts) - may be a bit out of date, and that these Control shortcuts are not global, they only work after clicking on the console.

It looks like Select-All & Copy have the shortcuts built-in by Qt/Windows (v's specifically configured??), consider whether to specifically add the Control-A & C shortcut so it is displayed for accessibility, like in 2015 above?

Also it'd be nice for the Console right-click menu to underline A in Select-All (ie Select-&All), (like the C in Copy) -  (ditto in editor too, for consistency). ATM it behaves like it is underlined, ie `r-click` `a` selects it.

Make Clear `C&lear` with Control-L shortcut. But I'm wondering from a general UI perspective, if it would be useful to have a global shortcut to clear the console?? But we are running out of keys...

With `Save As`, given the same menu item on the File menu, perhaps it should be `&Save Console As` (which matches the window title that opens) with Control-S shortcut. [_being picky_ - for consistency with other pop-ups, maybe the window title should be just 'Save Console' {note caps}, as 'Save console contents' is a bit redundent]

Repeatedly scraping the console on a timer is not ideal. Having a better means to determine that it either is, still-processing, or has-just-finished-processing would be helpful. I'm still pondering this.

MichaelAtOz, 2019-03-05:
I'd like to realign these console messages for consistent parsing of the console output.
I can probably do it via editing the file here, but wanted to see if anyone has an objection.
Anyone care??
https://github.com/openscad/openscad/blob/e827de293ba6f47beb8c4a746ad11b1fa2094157/src/mainwin.cc#L1390
just before
https://github.com/openscad/openscad/blob/e827de293ba6f47beb8c4a746ad11b1fa2094157/src/mainwin.cc#L1388

https://github.com/openscad/openscad/blob/e827de293ba6f47beb8c4a746ad11b1fa2094157/src/mainwin.cc#L2262
just before
https://github.com/openscad/openscad/blob/e827de293ba6f47beb8c4a746ad11b1fa2094157/src/mainwin.cc#L2292

F5
```
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 1
Geometry cache size in bytes: 728
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized CSG tree has 80200 elements
Total rendering time: 0 hours, 0 minutes, 5 seconds
Compile and preview finished.
```
F6
```
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
ECHO: "Processing: 15"
ECHO: "end"
Rendering Polygon Mesh using CGAL...
Geometries in cache: 124
Geometry cache size in bytes: 89544
CGAL Polyhedrons in cache: 17
CGAL cache size in bytes: 418640
Top level object is a 3D object:
Simple: yes
Vertices: 64
Halfedges: 192
Edges: 96
Halffacets: 68
Facets: 34
Volumes: 2
Total rendering time: 0 hours, 0 minutes, 5 seconds
Rendering finished.
```

MichaelAtOz, 2019-03-07:
Export/Image does not log to the console. So ATM I'm just sticking in an arbitrary timed wait.
For consistency it should log a similar message as all the other exports 'Image export finished: <file>'.

MichaelAtOz, 2019-03-07:
...and while on the subject, Design/flush-caches should log something, as it can run for an indeterminate time, and so the log history can make sense when comparing stats.

kintel, 2019-03-07:
I wonder if there are alternative query mechanisms to using a lot of extra logging. E.g. would it be possible to query the app for when the event loop starts running again?

MichaelAtOz, 2019-03-07:
Unfortunately the event loop (for preview/render/export) keeps running handing GUI events.
For some things Windows detects a long uninterrupted process by flagging the Title 'Not Responding' but it doesn't seem to be consistent or predictable.
I think the two cases, image export & flush, are the only two cases where other GUI properties can't be used. I think the image export should be logged just to be consistent with all the other exports, and a one-liner 'Cache flushed.' isn't a 'lot of extra logging' surely?

MichaelAtOz, 2019-03-07:
I found [this old post](https://github.com/openscad/openscad/issues/697), where OpenSCAD 2014 (presumably Qt) presented more information to Autohotkey, that disappeared in 2015. If a Qt can be wrangled to produce a 'readable' statusbar it could be used to pass 'status', e.g. pre/append 'Render Complete' or maybe 'Ready' <- after completing something. I'm looking at Qt accessibility, there is a Autohotkey library for accessibility (that I haven't looked at yet).

```
>>>>( TitleMatchMode=slow Visible Text )<<<<
statusbar
Viewport: translate = [ 0.00 0.00 0.00 ], rotate = [ 55.00 0.00 25.00 ], distance = 500.00
```

MichaelAtOz, 2019-03-07:
The lib can get what accessible info is there. (plus access to the console without clipboard)
[image]
So how about some signal in the statusbar?

MichaelAtOz, 2019-03-11:
I have used the library acc.ahk to get the console from accessibility info, so the op stuff about the console context menu is not need for this activity ATM*, it would tidy up the UI tho.
\* the Clear menu item may be useful in the future, having a shortcut would make life easier.

MichaelAtOz, 2019-03-26:
@nophead 
I just looked at the tests you updated.

There a a few of this style, e.g.
```
Test9: Missing include file appears
------
o rm missing.scad
o Open includemissing.scad
o Compile (F5)
o Verify that you get: WARNING: Can't open include file 'missing.scad'.
o echo "module missing() { sphere(10); }" >  missing.scad
o Reload and Compile (F4) - verify that the sphere appeared
o rm missing.scad
o Reload and Compile (F4) - verify that the sphere is still there
o echo "module missing() { sphere(20); }" >  missing.scad
o Reload and Compile (F4) - verify that the sphere increased in size
```
I presume you are not concerned with the geometry here, just that it finds & not-finds the file after F4.
So, replacing `module missing() { sphere(10); }` with `echo("hello world - missing.scad");` and check the console for that, would do? 
As that is something my GUI driver could do, along with other things, but it would be difficult to 'verify that the sphere is still there' - not necessarily impossibly tho.

As an aside
```
o rm missing.scad
o Reload and Compile (F4) - verify that the sphere is still there
```
Is that what you expect? Shouldn't it get a warning?

nophead, 2019-03-26:
Yes I think all the geometry could be replaced with echos, or even better do both.

You would get a warning with F5 but F4 doesn't actually do anything unless one of the files is updated.  F4 is what is called every 200ms when auto reload is on. Simply missing is ignored because the editor I use removes files before writing the new one. If a file goes missing OpenSCAD silently uses its cached version until it reappears or you force it with F5.

MichaelAtOz, 2019-04-02:
tl;dr - should we be releasing a new version without (or significantly reduced) functioning Accessibility?
@kintel @t-paul 
(yes I know the timing is bad)

My GUI driver prototype is working, but ATM requires the OpenSCAD window to be active/foreground.
Ideally it would run and be driven in the background.
I had difficulties with control limitations selecting menus in the background, so queried the AutoHotkey forum; tmplinshi gave me a solution using the [Accessibility](https://en.wikipedia.org/wiki/Microsoft_Active_Accessibility) library (which is what I was using to get the console).
He found the path to the menu hierarchy. This worked for 2015.03.
When I examined 2019.01-RC4 it lacked the menu hierarchy.
See [here](https://www.autohotkey.com/boards/viewtopic.php?f=76&t=63269&p=270943#p270943) for detail.
This means that menus would not be available to screen readers etc.
I'm thinking this is a bug, that newer Qt version may have changed re Accessibility.
While I obviously would like it for my purpose, should we be releasing a new version without (or significantly reduced) functioning Accessibility?
@kintel @t-paul 
(yes I know the timing is bad)

t-paul, 2019-04-03:
All of the core accessibility is handled by Qt, if that's not working well, there might not be much we can do. I do see menu access on Linux is working much better with the latest versions though.

MichaelAtOz, 2019-04-03:
#674 has some accessibility discussion. I went looking.

```
#if QT_VERSION < 0x050000
Q_IMPORT_PLUGIN(qtaccessiblewidgets)
#endif // QT_VERSION
```
https://github.com/openscad/openscad/blob/52c7ac3be59841c9dd1b1ea17340ae97de5d4fa4/src/openscad.cc#L527

I presume that says it's not using accessible widgets with the current Qt version.

MichaelAtOz, 2019-04-03:
But https://github.com/openscad/openscad/commit/ba8446e8bfbf770926a13b2d312187077adcb368 says it is built in with Qt-5.4.0 so that's not it.

MichaelAtOz, 2019-04-03:
FYI, I'm trying to understand what caused it to change from 2015.03, Qt version: 5.5.1.

ahh;
> 5.11 (5.11.3) 	New Accessibility Backend on Windows
> On Windows Desktop, the accessibility support was updated and vastly improved. The implementation is now based on Microsoft UI Automation (it was formerly based on Microsoft Active Accessibility).

vastly...

MichaelAtOz, 2019-04-03:
[Microsoft UI Automation](https://en.wikipedia.org/w/index.php?title=Microsoft_UI_Automation) 
> It also facilitates GUI test automation

[Qt 5.11 Brings New Accessibility Backend on Windows](https://resources.qt.io/development-topic-application-development/qt-5-11-brings-new-accessibility-backend-on-windows)
> Since the accessibility changes in Qt 5.11 are internal, existing accessible applications are not expected to require any changes to utilize the improved functionality provided by UI Automation. 

[qt docs](https://doc-snapshots.qt.io/qt5-5.11/accessible-qwidget.html)
> Qt already implements accessibility for its own widgets. We therefore recommend that you use Qt widgets where possible.
A QAccessibleInterface needs to be implemented for each element that you want to make available to accessibility clients.
...
> It is possible to implement interfaces by inheriting QAccessibleInterface and implementing its pure virtual functions.

I'm not Qt literate, but that sounds to me that it just needs to be turned on to get basic accessibility for standard widgets??