openscad/openscad #2846: Redefine vector using variables fails

idaho-1, 2019-03-05:
Using version 2015.03-3

// This seems to be a bug...
// Stephen Lewis 4-mar-2019
vvv = [0, 0, 0];
xxx = 123;
yyy = 456;
zzz = 789;
echo ("xxx is ", xxx, "yyy is ", yyy, "zzz is ", zzz);
vvv = [xxx, yyy, zzz];
echo ("vector is ", vvv);
// Surely vector should be [123, 456, 789]
// actually get [undef, undef, undef] (?)

Results:
Compiling design (CSG Tree generation)...
WARNING: Ignoring unknown variable 'xxx'. 
WARNING: Ignoring unknown variable 'yyy'. 
WARNING: Ignoring unknown variable 'zzz'. 
ECHO: "xxx is ", 123, "yyy is ", 456, "zzz is ", 789
ECHO: "vector is ", [undef, undef, undef]

thehans, 2019-03-05:
Redefining variables within a given scope is not valid, and newer versions produce a warning for that.

MichaelAtOz, 2019-03-05:
See [here](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/General#Variables) and [here](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/General#Variables_are_set_at_compile-time,_not_run-time).
Note you get 'Ignoring unknown variable' because when `vvv` is [instantiated](https://en.wikipedia.org/wiki/Instance_%28computer_science%29) (line 3), those variables do not exist. If you move `vvv = [0, 0, 0];` after `zzz = 789;` you don't get the warning, and you get the expected results:
```
ECHO: "xxx is ", 123, "yyy is ", 456, "zzz is ", 789
ECHO: "vector is ", [123, 456, 789]
```

idaho-1, 2019-03-05:
@thehans might produce warning but described in Wiki

// The value of 'a' reflects only the last set value
   a = 0;
   echo(a);  // 5
   a = 3;
   echo(a);  // 5
   a = 5;

idaho-1, 2019-03-05:
@MichaelAtOz OK, I got it now, thanks for explanation,
"counterintuitive" as they say...

MichaelAtOz, 2019-03-05:
Note that this redefinition, while it works, should not be used, it is a legacy of the -D var=val for command line usage. A better implementation of that in the future may mean redefinition could be outlawed.

closed 2019-03-05 by MichaelAtOz

nophead, 2019-03-05:
I hope not. This sudden move to outlaw things that have been used for a
decade is becoming a pain. How is one supposed to maintain code if the
language is forever changing in ways that are not backwards compatible?

t-paul, 2019-03-05:
I don't think `-Dvar=val` will go away as feature. What might become deprecated is specifying random non-parseable snippets (which are currently just appended to the existing script text). Also the implementation of how the value is set internally needs to change at some point.

MichaelPFrey, 2019-03-05:
> Note that this redefinition, while it works, should not be used, it is a legacy of the -D var=val for command line usage. A better implementation of that in the future may mean redefinition could be outlawed.

No, that is not true, at least not anymore. Those assigments are now marked so that the parser can handle that edge case separately. The much more important use case is, that you can include constants/defaults and overwrite them in your main file.

> (which are currently just appended to the existing script text).

This is not true anymore since #2460. Between the script and the ` -D var=val ` is now an End of Transmission sign
https://github.com/openscad/openscad/blob/8bdf072a465da409d5da4572b60b7892488f4d0e/src/ModuleCache.cc#L100
and this EoT sign
https://github.com/openscad/openscad/blob/8bdf072a465da409d5da4572b60b7892488f4d0e/src/parser.y#L197-L201
 is used to suppress the duplicate assignment warning:
https://github.com/openscad/openscad/blob/8bdf072a465da409d5da4572b60b7892488f4d0e/src/parser.y#L220-L222
that would otherways trigger:
https://github.com/openscad/openscad/blob/8bdf072a465da409d5da4572b60b7892488f4d0e/src/parser.y#L222-L224
It would be trivial to always warn on duplicate assignments or even stop parsing, but I consider including a file containing default model values to be a valid use case.