openscad/openscad #2847: Segmentation fault on rendering

plut, 2019-03-05:
openscad 2019.02.07, libcgal13 4.13-1+b2 (using the Debian repositories mentioned in http://www.openscad.org/downloads.html).

Running on the following file (which is a stupid model, granted) reliably produces a segfault when rendering, either in the GUI or in command-line via `openscad bug.txt -o bug.stl`.

[bug.txt](https://github.com/openscad/openscad/files/2932430/bug.txt)

Since the model is correctly previewed, might this be a CGAL bug instead?

t-paul, 2019-03-05:
The polyhedron is invalid (some faces are pointing in the wrong direction - [see this FAQ entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_model_showing_up_with_F5_but_not_F6?)). It should never crash OpenSCAD though.

kintel, 2019-03-07:
I can reproduce on macOS using CGAL 4.11. It works (reports error instead of crashing) with CGAL 4.8.1.

This was tested using old dev snapshots of OpenSCAD, so there's theoretically a possibility that we have code changes influencing this, but the stack trace points quite strongly to an uncaught (uncatchable?) issue inside CGAL.

kintel, 2019-03-07:
I suspect this to be the culprit: Someone commented out an assertion check in CGAL:

https://github.com/CGAL/cgal/commit/f1f82785acb767ed3f0c7918ab19b45771ec4a5d#diff-2d25c6c5eefcde7a443ef7269d33ee5eL210

kintel, 2019-03-07:
If someone can organize an upstream bug report that would be awesome

adrianVmariano, 2019-03-20:
I found that this code

    for (a=[0:120:240])
       rotate(a)
          polyhedron(points = [[0,.0911,0],[1,0,0],[0,-.0911,0],[0,0,.1]],
              faces = [[0,1,2], [1,2,3], [0,1,3], [0,3,2]]);

crashes  2019.01-RC3 under Windows when I attempt to render.  The first face of the polyhedron is backwards.  If I fix that it doesn't crash any more.  Is this the same bug?

MichaelAtOz, 2019-12-01:
Did anyone 'organize an upstream bug report'?

thehans, 2019-12-16:
I have now ^^^

> Did anyone 'organize an upstream bug report'?

thehans, 2019-12-16:
Well, that didn't quite go as planned.  Already closed and completely ignored the issue of commented-out assertion which I described as carefully as I could.

They did mention possibly calling [orient_polygon_soup()](https://doc.cgal.org/latest/Polygon_mesh_processing/group__PMP__orientation__grp.html#gaadc5bac3072b11d2313e40043c729d0a) which maybe we can use to auto-fix some bad inputs like this.  

Still would be nice to have catchable exceptions over crashes.  I might try replying later when I've cooled off... not happy about how that was handled.

MichaelAtOz, 2019-12-24:
If this seems related to the CGAL assertion being commented out, why...
With 2019.05 (with CGAL 4.11) I get
ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_external_structure.h Line: 1152
But with 20119.11.20.ci3989 (also with CGAL 4.11) I get a crash?
Same CGAL, so it must be what it is being fed??

thehans, 2019-12-24:
You sure your 2019.05 shows CGAL 4.11 under Library Info?  
For my Linux 2019.05 AppImage it shows 4.7

MichaelAtOz, 2019-12-24:
OpenSCAD Version: 2019.05
System information: Windows(TM) 6.1 SP 1.0 NTW 1 MSDN 724833 6 CPUs 6.06 GB RAM
User Agent: OpenSCAD/2019.05 (Windows(TM) 6.1 SP 1.0 NTW 1 MSDN 724833)
Compiler: GCC "7.3.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_60
Eigen version: 3.2.5
CGAL version, kernels: 4.11, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.11.2
QScintilla version: 2.10.4

Complete info;
[2019.05.lib.info.txt](https://github.com/openscad/openscad/files/3997298/2019.05.lib.info.txt)

MichaelAtOz, 2019-12-24:
Tho mine is probably a separate crash, both the OP code and @adrianVmariano's crash both 2019.05 & 2019.11.20  (CGAL 4.11), with Exception Code:	c0000005

Mine gets Exception Code:	40000015
Using
```
for (i = [0:45:360])
  rotate ([0,0,i])
    translate ([10,0,0])
      import("rotate201905-rot2.stl");
```
with this STL created with code below, (similar to the earlier one, Netfabb shows SI's, Meshlab doesn't)
[rotate201905-rot2.zip](https://github.com/openscad/openscad/files/3997329/rotate201905-rot2.zip)

```
difference (){
  translate ([5,-2,0])
    cube ([20,4,4]);
  translate ([0,0,2])
    rotate(2)
      rotate_extrude(angle=360,$fn=60)
        translate ([11,0,0])
          circle (d=3.4);}
```
Sorry probably a red herring.

thehans, 2019-12-24:
Please keep comments on-topic to the original test cases.  Its getting very hard to keep track of these mixed up issues.  

The OP's case *doesn't* crash on 4.7 (nor 4.8 according to kintel), and it *does* crash on 4.11, regardless if built from 2019.05 tag or latest master.

MichaelAtOz, 2019-12-24:
For completeness, 2019.11.24.nightly on Debian has:
CGAL version, kernels: 4.9, Cartesian, Extended_cartesian, Epeck
and does NOT crash with the rotate...stl.
With the OP scad it gets
```
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: pe != 0 File: /usr/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h Line: 212
```
With adrianVmariano's it doesn't get an error and renders something like a 3 point star.

thehans, 2020-12-10:
I realized there were already some checks for whether a mesh is closed, and valid, but they were only used in debug messages.
I made a PR #3526, which stops trying to convert to Nef if either of those checks return false.

There is still some room for improvement, but at least I think this will avoid segfaults.
 - Error messages don't have access to source Location info.  Since its converting from a Polyset, maybe we should have Location info stored in each of those upon creation?  
 - CGAL's orient_polygon_soup might still be able to correct for some of these class of errors, but I was somewhat confused on how to best implement it.  Our current code converts `Polyset` to `Polyhedron_3`, and from there to Nef.  But Polyhedron_3 doesn't appear to have direct access to underlying points and faces that `orient_polygon_soup` would need.

t-paul, 2020-12-18:
Fixed via #3526.

closed 2020-12-18 by t-paul