openscad/openscad #2849: AST dump does not handle "use" command

thehans, 2019-03-05:
We already have a test for this, but the "expected" results are just to not show the used library code.
`astdumptest_use-tests`, from `testdata/scad/functions/use-tests.scad` / `testdata/scad/templates/use-tests-template.scad`

kintel, 2019-03-05:
See also #1743, which is a bit stalled

thehans, 2019-03-05:
I guess one issue with implementing this is that there doesn't seem to be a way to represent how use's are evaluated within the current language.

blah.scad
```
use <blah2.scad>

echo(is_undef(x));
blah();
```
blah2.scad
```
  x = 10;
  module blah() {
    cube(x);
  }
```

So blah.scad should `ECHO: true`, But the module call can use the variable still.

I thought maybe I could test some trick with an anonymous block scope, but it seems to just evaluate as if the block wasn't there.  So proposed AST format would be something like this:
```
/*use <blah2.scad>*/ {
  x = 10;
  module blah() {
    cube(x);
  }
}

echo(is_undef(x));
blah();
```
It outputs `ECHO: false`, it still sees the variable, and it is able to call the module.  I honestly expected the `blah();` call to not be accessible either in this case.  I wonder if changing this behavior to make variables local to a block like this, but continue to "export" module and function definitions would be a sensible way to handle it?

Of course this could affect backwards compatibility if scripts are using blocks like this for some reason.

MichaelAtOz, 2019-03-05:
> if scripts are using blocks like this for some reason

This is come up before. 
They are used, including by me, mainly for convenience but I would expect it is very rare, I only recall it in one other case some time ago.
I'd think that backwards compatibility for such intricacies could be sacrificed for the greater good.

thehans, 2019-03-05:
Hmm after some more thought, it also needs to ignore any module instantiations within the use block.    
I think maybe the proper way to handle it for backwards compatibility would be to define a new type of "use" node, that evaluates its contents in exactly the same way.  The filename parameter would just be for reference, not affecting what it evaluates, so you can add stuff like this to the use'd library:
```
use("blah2.scad") {
  x = 10;
  module blah() {
    cube(x);
  }
  echo("hi");
  assert(false);
  sphere(100);
}
```

kintel, 2019-03-06:
I agree that the swallowing of anonymous blocks is annoying and I think it was "designed" that way by accident. If we can make a good case for killing it, I'm all for it. I'm just uncertain how many people depend on that feature.