openscad/openscad #2852: OpenSCAD version 2019.01-RC2 hangs during preview or render

CodeBendor, 2019-03-09:
In OpenSCAD version 2019.01-RC2 (git 35858ead) the following code hangs the program:

    module ConcaveCylinder(length, radius, smooth) {
        rad = radius - smooth;
        minkowski() {
            difference() {
                cylinder(length,r=rad,$fn=50);
                union() {
                    translate([0,0,length])sphere(rad,$fn=50);
                    sphere(rad,$fn=50);
                }
            }
            sphere(smooth,$fn=50);
        }
    }
    ConcaveCylinder(30,5,1);

t-paul, 2019-03-09:
Using `minkowski()` with 2 very high vertex count (and worst case concave) objects it just taking very long and may error out due to exploding memory requirements.

closed 2019-03-09 by t-paul

MichaelAtOz, 2019-03-11:
FWIW The facet count of the 2nd minkowski object seems to be the memory consumer.

As above it went to 6+GB working set and crashed when commit size hit 16GB, after a very long time.

```
module ConcaveCylinder(length, radius, smooth) {
  rad = radius - smooth;
  fn=32; fn2=16;
  minkowski() 
  {
        difference() {
            cylinder(length,r=rad,$fn=fn);
            union() {
                translate([0,0,length])sphere(rad,$fn=fn);
                sphere(rad,$fn=fn);
            }
        }
        sphere(smooth,$fn=fn2);
    }
}
ConcaveCylinder(30,5,1);
```
Ran for an hour, used 2.8GB memory.