openscad/openscad #2857: MSVC Build finishing touches

thehans, 2019-03-10:
This issue is for tracking remaining tasks to make the MSVC build first-class.

See original issue https://github.com/openscad/openscad/issues/2814 for build instructions.

- [ ] Create VCPKG port for Lib3MF so we can start enabling that in MSVC builds.  I created a request for this but have not worked on it yet: https://github.com/Microsoft/vcpkg/issues/5482  I think it should be fairly straightforward as the project already uses cmake.
- [ ] MSVC build is not copying any of the subdirectories into the build output path: "colorschemes", "examples", etc.
- [ ] No build for console application `openscad.com`.  Needs its own CMakeLists.txt
- [ ] Need to get `tests\CMakeLists.txt` running.  I'm not sure how to set up multiple CMakeLists.txt in Visual Studio.  May need to use cmake [add_subdirectory](https://cmake.org/cmake/help/v3.8/command/add_subdirectory.html) so that child projects are included from the main one (including winconsole, see above).
- [ ] It would help to wrap the vcpkg setup into a script(`.bat` probably) file. Currently building from within VS GUI, would be nice to work out a full command line build that can be run on AppVeyor possibly.
- [ ] Main CMakeLists.txt could probably still use some cleanup/refactoring.  It works, but seems a little messy, maybe splitting out some of the windows-specific stuff into an included cmake file.
- [ ] Move build instructions to wiki

nophead, 2019-03-10:
As a matter of interest are all the things that get built for the test actually still used? Don't the tests just use the OpenSCAD executable in command line mode with Python scripts?

thehans, 2019-03-10:
The tests makelist no longer builds any executables as far as I know except for diffpng in the case where imagemagick can't be found.  It mainly sets up ctest and generates a few template based tests.  

The main problem is that it still checks for a bunch of packages as if it's going to build those old binaries.

nophead, 2019-03-10:
Is that a recent change. The last time I ran the tests on MSYS2 a few weeks
ago it compiled a lot of OpenSCAD source files to make other programs as
far as I can tell.

thehans, 2019-03-10:
I'm pretty sure its been that way for a few months at least, not sure exactly when it changed.  I think @t-paul changed it at some point.

nophead, 2019-03-10:
I just built with MSYS2 after getting a new gcc that seems to have fixed
the regression that broke QT, and yes it no longer builds all the other
test programs, great.

roel-v, 2024-02-01:
Does this still work for anyone? I've been trying to get this to work (all from the command line), and my first issue was that there is no unofficial-fontconfig package in vcpkg. So I changed that to Fontconfig (note the capital) which I think should work but didn't. So I commented it out for now. Then I got to a problem with gettext, saying it couldn't be found. So I changed find_library to find_package for gettext (suggested online somewhere), but then I got this error instead:

CMake Error at D:/vcpkg/scripts/buildsystems/vcpkg.cmake:859 (_find_package):
  _find_package called with invalid argument "libintl"
Call Stack (most recent call first):
  CMakeLists.txt:289 (find_package)

Finally (as far as I remember), I also had an issue with qscintilla depending on qt6. But I don't think it actually does, from what I understand, qscintilla works with both qt5 and 6. I changed cmake\Modules\FindQt5QScintilla.cmake by adding qscintilla2_qt6 to NAMES of find_library, as I think it's just a name and it will actually use qt5. But that's just a theory as I haven't gotten to a stage yet where I actually have working build files, let alone that it compiled/linked, due to the first problems mentioned.

Does any of this ring a bell before I dive deeper?

jordanbrown0, 2024-02-02:
The most recent MSVC work that I'm aware of is PR #4126.