openscad/openscad #2859: OS X 10.14.2 [objects/lexer.cxx] Error 1

ScaredyCat, 2019-03-10:
I'm not sure where to go with this one, a general error that I can't find anything to give me hints at what's causing this. Any clues?

[Makefile.txt](https://github.com/openscad/openscad/files/2949565/Makefile.txt)

> $ qmake openscad.pro
> Project MESSAGE: If you're building a development binary, consider adding CONFIG+=experimental
> Project MESSAGE: Using ccache
> Project MESSAGE: Using QOpenGLWidget
> Project MESSAGE: Using libc++11
> Project MESSAGE: Using C++11
> Project MESSAGE: 3MF Import/Export disabled
> gort:openscad$ make
> flex -o objects/lexer.cxx --header-file=objects/lexer.hxx src/lexer.l
> make: *** [objects/lexer.cxx] Error 1

t-paul, 2019-03-10:
Check where that `flex` binary is coming from or if there are multiple versions installed? Maybe it's calling one from some older install which is now confused by changed shared libs or something.

thehans, 2019-03-10:
Maybe try a `make clean` then make sure objects dir is empty before running qmake then make again.

kintel, 2019-03-10:
Also, there's a difference between `lexer.l` and `comment_lexer.l` - make sure you test the right thing

kintel, 2019-03-10:
Also, where did you get flex? `flex version 2.5.37`
I've got `flex version 2.5.35 Apple(flex-31)`

ScaredyCat, 2019-03-10:
Ok, I had a version here:

> gort:openscad$ which flex
> /opt/local/bin/flex

So onvestigated, looks like it came from an old macports install. 
> gort:openscad$ which flex
> /usr/bin/flex

> gort:openscad$ flex -V
> flex 2.5.35 Apple(flex-31)

I haven't used macports for a very long time, so much so that I had to upgrade for it to even work. Part of htat process involved uninstalling everything that macports installed - which fixed this issue. 

Thanks for the heads up on the version numbers, I'd totally forgotten about ports being installed :/ - sorry for the error on my part.

closed 2019-03-10 by ScaredyCat