openscad/openscad #2860: /etc/apt/sources.list.d/opencad-nightly.list offers arm64 packages even with [arch:amd64]

Philius, 2019-03-10:
Days ago I did a fresh install of Debian Stretch amd64.
The last step was to follow [the openscad nightly instructions](http://www.openscad.org/downloads.html) which I did.
But when it came time to install, the synaptic package manager said its dependencies weren't installable.
It took me a while to figure out the problem.
Wrong distribution? Backports? No it was because the packages offered were (according to synaptic) for the arm64 architecture.
A bit of googling later and I found [how to get rid of arm64 on apt](https://askubuntu.com/questions/917081/how-to-get-rid-of-arm64-in-apt) which didn't work!!!
So I just downloaded the openscad-nightly_20190309T132651.git40379b3f5+2956.1_amd64.deb and installed it with gdebi.
It's still showing up as arm64 but now with the native package installed too.
My hunch would be that it's an OpenSuse configuration or packaging issue.

t-paul, 2019-03-10:
Please show the content of `/etc/apt/sources.list.d/opencad-nightly.list`.

Philius, 2019-03-11:
I can't attach the file without zipping it but its contents are:
`deb [arch=amd64] http://download.opensuse.org/repositories/home:/t-paul/Debian_9.0/ ./
`
More curious is that with openscad installed from a .deb file with gdebi, apt and synaptic offer an upgrade to the latest amd64 version even though a search in synaptic only shows the current amd64 version and the arm64 version and its debug symbols package.

Removing openscad-nightly.list also causes the arm64 architecture to go away.

t-paul, 2019-03-11:
Snippet is fine, thanks (you can just add `.txt` to files to upload, I think).

That look totally normal, I'm not sure what's going on. I'll ask around with some of the Debian people. The OBS repo does have the ARM stuff too, but I would not have expected that to trigger strange behavior.
Does that happen also on command line with `apt`, maybe it's `synaptic` being a bit too eager to install stuff?

t-paul, 2019-03-11:
Oh, could you please copy&paste the output of `apt-cache policy openscad-nightly`? Not sure if it helps, but sometimes it gives some useful extra info.

Philius, 2019-03-11:
`# apt-cache policy openscad-nightly`
`openscad-nightly:`
`  Installed: 20190310T214623.git394d147a4+2987.1`
`  Candidate: 20190310T214623.git394d147a4+2987.1`
`  Version table:`
` *** 20190310T214623.git394d147a4+2987.1 500`
`        500 http://download.opensuse.org/repositories/home:/t-paul/Debian_9.0 ./ Packages`
`        100 /var/lib/dpkg/status`

`# dpkg --print-architecture`
`amd64`

`# dpkg --print-foreign-architectures`
`i386`

Philius, 2019-03-11:
`# dpkg -l '*openscad-nightly*'`
`Desired=Unknown/Install/Remove/Purge/Hold`
`| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend`
`|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)`
`||/ Name                  Version         Architecture    Description`
`+++-=====================-===============-===============-===============================================`
`ii  openscad-nightly      20190310T214623 amd64           script file based graphical CAD environment`

The synaptic package manager lists this along with the arm64 version and the debugging symbols package for arm64.