openscad/openscad #2862: Make the CI publish artifacts

KOLANICH, 2019-03-11:


thehans, 2019-03-11:
For which CI Server?  CircleCI has downloable artifacts already e.g. https://circleci.com/gh/openscad/openscad/2287#artifacts/containers/0

KOLANICH, 2019-03-11:
>Which CI Server

Appveyor. CircleCI is not privacy friendly. They demand to enable JS, but have a lot of tracking scripts. It may make sense to move all the builds to AppVeyor or maybe even to GitLab.

t-paul, 2019-03-11:
All available artifacts are on CircleCI (Windows + Linux AppImage). AppVeyor is not meant to produce downloadable artifacts, it's used for running the test suite.

We can't move everything to AppVeyor, it's way too slow for that (at least with the free plan we are using).

You can get the URLs of the downloads by using the CircleCI API:
`https://circleci.com/api/v1.1/project/github/openscad/openscad/${build_number}/artifacts`

Where `${build_number}` is the value shown in the "details" link of the build checks.

closed 2019-03-11 by t-paul

t-paul, 2019-03-11:
More details on the API can be found in the script [circleci-download-artifacts.py](https://github.com/openscad/openscad/blob/master/scripts/circleci-download-artifacts.py)

It's only tested on Linux, but as it's only using pretty normal stuff, it might run on Windows aswell. It simply outputs the URLs of the latest `master` in case there are new builds, so on Linux it's just used as `wget $(./circleci-download-artifacts.py)`.