openscad/openscad #2863: Lint flag

Maxattax97, 2019-03-12:
I'm working on a large library, and was wondering if a simple flag could be provided to check if all the syntax in a file is valid/can be rendered.

At the moment, I [adapted some code](https://github.com/Maxattax97/cornucopia/blob/master/scripts/unit_test.py#L78) from MCAD to simply try and compile the file to STL, however this has some drawbacks. Notably, I have to add some geometry so the test doesn't fail when I have just functions in the file that render nothing. Another problem that could come up is geometry which has overlapping faces with the cube I've inserted.

To add onto this, I have a question; I'd like to find a way to format OpenSCAD source files, since when I look at other repo's I quickly find that the code doesn't follow a consistent style. I'm not asking for that to be added to this project, but I wanted to pick your brains on what you think might be the easiest way to get this done. I know that [some](https://github.com/openscad/openscad/blob/master/src/parser.y) grammar [files](https://github.com/openscad/openscad/blob/master/src/comment_parser.y) exist, so there's a possibility I could reconstruct the source from the AST, but I'm not sure that's a simple task exactly. I've had limited success with C-style formatters like Uncrustify.

t-paul, 2019-03-12:
Putting all questions into a single issue makes it a bit hard to handle...

For syntax validation `openscad CSG.scad -o CSG.ast` might be relatively near to what you are asking. It's parsing and dumping the AST. Right now the output file must have the `.ast` suffix to define the output type, so you can't just use `/dev/null` for that.

This could even be also the answer for for the second question, but that needs work as the input -> AST transformation does not preserve all information yet (e.g. comments).

nophead, 2019-03-12:
Every preview or render always checks all of the syntax whether it is used or not. What you are describing are semantic errors, such as unused variables and bad geometry creation and those only become apparent when code is used.

To find most of those I have test files for all of my library classes, which instantiate examples of all objects of that type and it renders a picture that I can visually inspect plus I see any errors or warnings in the console. I also add a few asserts to trap parameter values that would generate bad geometry but generally I don't care if garbage in gives garbage out. 

For example here is the test for my D-connector class.
```
module d_connectors()
    for(socket = [false, true])
        translate([socket ? len(d_connectors) * (d_flange_width(d_connectors[0]) + 10) : 0, 0])
            layout([for(d = d_connectors) d_flange_width(d)], 10)
                rotate(90) {
                    d_plug(d_connectors[$i], socket, pcb = $i == 2, idc = $i == 1);

                    if(socket)
                        d_connector_holes(d_connectors[$i])
                            d_pillar(d_connectors[$i]);
                }

if($preview)
    d_connectors();
```
And here is what it produces when I run my test script.
[image]

Since my library is primarily for generating vitamins to place in my assembly views the utilities it has are mainly to support that, so instantiating all the objects also tests utilities like sweep indirectly.

Maxattax97, 2019-03-13:
@t-paul That AST conversion is very neat and helpful; I've implemented it into my unit testing.

I've also finished an [OpenSCAD source formatter](https://github.com/Maxattax97/openscad-format) today, if anyone here is interested. It's working but I haven't bumped to a release version yet since I'm still working out bugs here and there.

You can install it like so: `npm install -g openscad-format`
And use it like so: `openscad-format -i './**/*.scad'`
Issues and PR's are welcome.

Thanks for your help guys, I'll mark this as solved!

closed 2019-03-13 by Maxattax97