openscad/openscad #2865: Change CFLAGS

baerbock, 2019-03-13:
I don't redistribute my binaries, so I would like to compile openscad with

`ccache g++ -c -pipe -DSTACKSIZE=8388608 -fno-strict-aliasing -std=c++11 -O2 -march=native -mtune=native -O3 `
instead of
`ccache g++ -c -pipe -DSTACKSIZE=8388608 -fno-strict-aliasing -std=c++11 -O2 -march=x86-64 -mtune=generic -O2 `

Where are the CFLAGS for Linux defined? I could only find them for OS X.

Thanks in advance!

kintel, 2019-03-13:
Try `QMAKE_CXXFLAGS += -O3`

closed 2019-03-13 by kintel

baerbock, 2019-03-13:
@kintel Does `-O3` in a gcc command override a `-O2` in the same line?

kintel, 2019-03-13:
I'm not sure. Both gcc's and qmake's behavior should be relatively well documented though. Best way is probably to try it.

thehans, 2019-03-13:
https://gcc.gnu.org/onlinedocs/gcc-5.2.0/gcc/Optimize-Options.html

    If you use multiple -O options, with or without level numbers, the last such option is the one that is effective.

thehans, 2019-03-13:
Also I think you can remove unwanted flags similarly: `QMAKE_CXXFLAGS -= -O2`