openscad/openscad #2866: QCoreApplication::postEvent: Unexpected null receiver

thehans, 2019-03-14:
I see this message from Qt pretty much every time I close OpenSCAD when it was started at command prompt on linux mint 19.  

Don't know if there's any way to debug, there's no exception to catch or anything, just a slightly annoying message.

kintel, 2019-03-14:
It should be possible to get a stack trace if you build with address sanitation, but I haven't tried that myself

kintel, 2019-03-14:
It was also mentioned in #2393 as a possible Qt 5.10 issue

thehans, 2019-03-14:
Yeah it seems to be related to the QSound::play call, and the bug linked from that discussion:
https://bugreports.qt.io/browse/QTBUG-66910

It only happens if I do a render.  If I build with the QSound call commented out it doesn't happen.

I have 5.9.5, looks like the fix applies to 5.9.7.  I'll just close this as its an upstream issue and pretty inconsequential.

closed 2019-03-14 by thehans