openscad/openscad #2868: Bad geometry - holes & degenerates

MichaelAtOz, 2019-03-15:
This is with RC3, it is worse than with 2015.
Starting from where I noticed a problem,  looks  very bad.
[image]
In a nut shell that is a result of the following, the minkowski takes an hour, so I'll save you from the full code.
```
minkowski() scale([0.15,0.15,1]) linear_extrude(height=1.5, convexity=16) offset(-0.75) union() difference() union() text("𒄉", size=96, font="Akkadian:style=Regular",$fn=36);
```
Reducing it finds scale introduces degenerate faces: (note this is inverted compared to above)
[image]

The concentration of 'bad' in the top image correlates to the degenerations.
So I presume it is 'expected' that the further processing of the minkowski() will cause the nastiness??

I went back to 2015.03, it has 1/3rd of the degenerations than RC3, just those at the lips:
[image]

So, is it concerning that RC3 is worse than 2015? (note I haven't tried the full minkowski on 2015 yet)

If you want to replicate exactly, you'll need [the font](https://fontlibrary.org/en/font/akkadian), &
```
$fs=1.25; 
$fa=3;
//minkowski() 
  {
   scale([0.15,0.15,1])  
      linear_extrude(height=1.5, convexity=16) {
        offset(-0.75)
          union() {
            difference() {
              union() {
                text("𒄉", size=96, font="Akkadian:style=Regular",$fn=36);
                translate([105,83,0])
                  circle(d=12);
                translate([153,57,0])
                  circle(d=13);
              }
              translate([105,84,0])
                #circle(d=6);
              translate([153,57,0])
                #circle(d=5);
            }
            translate([190,36,0])
              circle(d=10);
            translate([190,44,0])
              circle(d=10);
            }
        }
    
  sphere(0.25,$fn=18);
  }
```

BTW if I change the order to scale([0.15,0.15,1]) minkowski() ... I get this (reused the cache as it ran in 0s)
[image]

So something funny in there somewhere...

MichaelAtOz, 2019-03-15:
Actually it looks like offset() may be the cause.
```
//minkowski() 
  {
   scale([0.15,0.15,1])  
      linear_extrude(height=1.5, convexity=16) 
    {
        //offset(-0.75)
          union() {
...
```
Is clean:
[image]

If I tram the lip circles to change the angle of incidence (re adding the offset) it goes away:
[image]
and the other two locations are no longer degenerate.

MichaelAtOz, 2019-03-15:
AAArrrgh! The second above was without scale(), re-add that and those other two locations reappear:
[image]
Offset & scale are presumably having floating point differences of opinion.

MichaelAtOz, 2019-03-15:
The full minkowski in 2015, ie the same as the very first image, no holes, so RC3 is producing worse output than 2015.
[image]

MichaelAtOz, 2019-03-15:
With a positive offset() it produces different degenerations.
```
//minkowski() 
  {
   scale([0.15,0.15,1])  
      linear_extrude(height=1.5, convexity=16) {
        offset(0.5)
          union() {
```
[image]

Again only with both scale() & offset(), either by themselves doesn't degenerate.

kintel, 2019-03-15:
If you have a good workflow for detecting this, would it be possible to find the smallest possible example using a more common font?
Alternatively, reproduce this by exporting to DXF/SVG or a 2D polygon and making it break again on import?

MichaelAtOz, 2019-03-15:
```
$fs=1.25; 
$fa=3;
   scale([0.15,0.15,1])  
      linear_extrude(height=2, convexity=16) 
        offset(-0.75)
          import("babel-fish.dxf");
```
[babel-fish.dxf.txt](https://github.com/openscad/openscad/files/2973101/babel-fish.dxf.txt)

As above:
[image]
Offset(+0.75):
[image]

MichaelAtOz, 2019-03-16:
With a built in font. I chose those glyphs semi-randomly with complex angles & some holes.
```
$fs=1.25; 
$fa=3;
   scale([0.15,0.15,1])  
      linear_extrude(height=2, convexity=16) {
        offset(-0.75)
          text("₡₢₥₦₩₯₱₳", font="Liberation Serif:style=Regular", size=96, spacing=0.75 ,$fn=36);
      }
```
[image]

Simpler, just "₯₱"
[image]

nophead, 2019-03-30:
Does it work if you do the scaling in 2D before the linear extrude?

MichaelAtOz, 2019-03-30:
No.

MichaelAtOz, 2019-04-08:
#2915 is producing degenerate faces too, and involving the 'small number' problem.
On reflection this, babel-fish/text, issue may be 'small number' problems too.
I'm wondering whether offset() brings in another floating point operation? Yes, Clipper.
With the intricate curvature of glyphs, points being minutely separated, floating point producing some equivalent to ['Brownian motion'](https://en.wikipedia.org/wiki/Brownian_motion). 
I notice CLIPPER_SCALE = 65536, as Clipper is Integer based, it converts both ways introducing more uncertainty. I'm wondering if that and the GRID is causing noise?
Perhaps that is the point to do some mesh repair?
Where are the Heisenberg compensators :/

nophead, 2019-04-08:
Clipper itself is supposed to be very robust with respect to self intersections, etc, and I don't think I have seen problems in 2D. I have used offset(0.01) to clean up a bad SVG before linear extruding it.

Yes any snap to grid or truncation to lower precision will break things. OpenSCAD does that all over the place with int64 to double, double to float, rational to double and some explicit grid snaps, which can be finer than the resolution of float as you move away from the origin.

donbright, 2019-05-02:
Clipper also does a large amount of converting from Integer to Double and back again, although it is very robust in general.  But Clipper converting to Int64 is essentially a form of "grid snap".

OpenSCADs floating point grid size could, in theory, be settable at runtime, so that these types of tests could be done by ordinary users to debug the engine better, without having to try to compile the program themselves with a modified gridsize. Also its called a grid but its actually not a "regular grid" so much as a "rectangular mesh" (since its based on floating point binary numbers, which do not form a regular grid  in space).

Different OpenSCAD-grid sizes, in and of themselves, will produce a performance change of maybe 10-20% depending - this is because the Rational Number math of CGAL, well, when you use a grid size that equates to an exact fraction of a binary number, the Double->Rational conversion produces a fewer number of digits in the resulting Rational denominator and numerator. This is expressed in the Bignum routines of MPFR/GMP taking less time to walk through the 'legs' of the Big Integers that  underly their Rational number math.

MichaelAtOz, 2020-12-15:
I'm revisiting some issues with 2020.12.RC2.

This one seems fixed.
It would be interesting to have conjecture on why?

The non-minkowski version above had zero self-intersections.
Zero with minkowski too:
[image]

Tho apparently double render time, I will need to confirm this. So not closing the issue yet.

thehans, 2020-12-31:
> This one seems fixed.
> It would be interesting to have conjecture on why?

Possibly from changes to how large text is scaled (#3263)
Previous code had limited precision to glyph scaling, which may have led to some self-intersection/degenerate polygon edges etc.

kintel, 2024-03-30:
Closing as this seems to be fixed and probably not worth chasing why

closed 2024-03-30 by kintel