openscad/openscad #2871: Allow console window behind others

MichaelAtOz, 2019-03-16:
I wanted to have the detached console just poking out the side of the editor/preview.
But console stays in front when the other window is active.
[image]

css911, 2019-03-16:
@MichaelAtOz, Hi... I Chetan Shinde, a gsoc aspirant can I work on this issue?

MichaelAtOz, 2019-03-16:
@kintel ?

t-paul, 2019-03-16:
I'm not sure that is possible. This is how Qt handles the dockable windows. Maybe there's some new stuff in Qt5 we can investigate after dropping Qt4.

kintel, 2019-03-16:
If you find a way, sure, but this is likely to require significant knowledge of Qt.

RomitKumar, 2019-04-11:
Resolved at PR #2930

MichaelAtOz, 2019-04-11:
@RomitKumar Thanks.

closed 2019-07-07 by t-paul