openscad/openscad #2877: Add command line option to specify the export format

t-paul, 2019-03-17:
Currently the export format is determined by looking at the extension of the file given to the `-o` option (e.g. `-o export.stl` will set the format to STL).

It would be useful to have an additional parameter that can force a specific export format in cases where the file name can't have the needed extension (e.g. when using fifos or pipes).

Suggestion for the option: `--export-format`

So if the option is given, the export format is forced ignoring the file name completely. Ideally the code should be changed to be a bit more driven by a declared list of acceptable format, removing the currently used long list of if/else constructs.

RomitKumar, 2019-03-20:
What should be done when both -o and --export-format option is used? Maybe report a error.

t-paul, 2019-03-20:
As mentioned in the last paragraph, I think that giving the options should always force what format is selected regardless of how the file is named. I'm not sure if it makes sense to add a warning in case of obvious collisions, but then the user is asking for this specifically, so I tend to say we don't want a warning.

RomitKumar, 2019-03-20:
So, when issuing commands like:
1. `openscad --export-format stl -o foo.png bar.scad`
A foo.stl file should be made
2. `openscad --export-format stl bar.scad`
A untitled.stl file should be made. 
3. `openscad --export-format stl -o foo.png bar.scad | MySTLFileProcessor`
A foo.stl file should be made and passed to MySTLFileProcessor
4. `openscad --export-format stl bar.scad | MySTLFileProcessor`
A untitled.stl file should be made and passed to MySTLFileProcessor. (And maybe untitled.stl should be removed or maybe no untitled.stl file is generated in the first place and the bit stream is directly passed to MySTLFileProcessor?)

t-paul, 2019-03-20:
1. If the user gives strange instructions we probably should still follow those, so that would actually create a STL file named `foo.png`
2. Good question, normally that would start the GUI as there is no `-o` given. So we can either continue doing that ignoring the `--export-format` or give an error for incompatible actions. I'm leaning towards the starting the GUI option.

I would not cover the pipe stuff just yet, there's a separate issue as followup. The common way to do this seems to be a special file name like `-o -`.

RomitKumar, 2019-03-22:
Made a fix for this issue at PR #2887

donbright, 2019-05-02:
IN my opinion this is bad UI. There are some basic guidelines that can be followed to determine answers to questions like this, some good old examples are here: 

https://modelessdesign.com/backdrop/401

Programs should not create unpredictable, nonsensical output, and also the program should provide feedback to the user when there is some input error that is most likely user error. It should behave in a manner most consistent with what the user expects based on common sense. Not allow what are most likely user mistakes to go through unreported.

For example there is almost no reason someone purposely wants to create a file named ending STL that is actually a PNG. What is more likely to happen, as most of us have done it many times, is someone accidentally makes a script that gives contradictory information, to export-format to STL but filename was accidentally left PNG, this creates a 'garbage PNG' file that is actually STL and then whatever program the system uses to open the "PNG" file either crashes or gives a mysterious error about not opening it. 

Also if someone is using the command line with command line options there is probably 0 percent chance they want the GUI to start. This just makes no sense to go to the trouble of typing "openscad -export format stl x.scad" and for it to pop up the GUI. They type in export, that obviously means they want to export a file, so it should be able to figure out the output name based on the .scad file name.

RomitKumar, 2019-05-02:
According to my PR, whenever --export-format argument is used, the format of the output file is as given in argument. Also the extension of output file is as per its format. So, on issuing following command:
`openscad --export-format stl -o foo.png bar.scad`
A stl file is created by the name of foo.stl

However, if the export format argument is not correct, the format as well as extension of file is derived from the extension given in -o argument.
`openscad --export-format st -o foo.png bar.scad`
This commands ignores export format argument st and creates a png file by the name of foo.png.

In any case, the file extension matches with file format. So there should be no problem of unwanted crashing.

Regarding `openscad --export-format stl bar.scad`, I think your suggestion is a good idea and seems more natural. I guess this should be implemented. What do you think @t-paul ?

t-paul, 2019-05-02:
Yes, I like the suggestion too. It allows for a simpler command line. However it is actually a bigger change as currently the "do command line, not GUI" is triggered by the `-o` options specifically which is a pretty clear statement. If we change that, we need to make sure it's not getting confusing in other combinations.

So we probably want to collect all the cases into a table or something to get an overview of all the scenarios.

RomitKumar, 2019-05-04:
The possible combination of commands and expected output:

1. `openscad -o foo.stl bar.scad`
Description: Only -o argument provided. Valid extension.
Output File Name: foo.stl
Output File Format: stl 

2. `openscad -o foo.st bar.scad`
Description: Only -o argument provided. Invalid extension.
Output File Name: _No output file_

3. `openscad --export-format stl bar.scad`
Description: Only --export-format argument provided. Valid extension.
Output File Name: bar.stl
Output File Format: stl 

4. `openscad --export-format st bar.scad`
Description: Only --export-format argument provided. Invalid extension.
Output File Name: _No output file_

5. `openscad --export-format stl -o foo.stl bar.scad`
Description: Extension in both the arguments are same and valid.
Output File Name: foo.stl
Output File Format: stl 

6. `openscad --export-format st -o foo.st bar.scad`
Description: Extension in both the arguments are same and invalid.
Output File Name: _No output file_

7. `openscad --export-format png -o foo.stl bar.scad`
Description: Extension in both the arguments are different. Both the arguments are valid.
Output File Name: foo.png
Output File Format: png

8. `openscad --export-format pn -o foo.st bar.scad`
Description: Extension in both the arguments are different. Both the arguments are invalid.
Output File Name: _No output file_

9. `openscad --export-format png -o foo.st bar.scad`
Description: Extension in both the arguments are different. --export-format argument is valid, -o argument is invalid.
Output File Name: foo.png
Output File Format: png

10. `openscad --export-format pn -o foo.stl bar.scad`
Description: Extension in both the arguments are different. --export-format argument is invalid, -o argument is valid.
Output File Name: foo.stl
Output File Format: stl

Current implementation in PR produces the same output as mentioned for all cases except for case 3 and 4, in the case of which gui is launched. In all other cases, gui is not launched.

RomitKumar, 2019-05-04:
I guess that expected output for case 3 and 4 should not cause any confusion.

t-paul, 2019-07-07:
Closed via #2903.

closed 2019-07-07 by t-paul