openscad/openscad #2878: Error/Warning shows wrong file name for unsaved buffer

t-paul, 2019-03-17:
If the buffer / main window is not yet associated with a file, some of the warnings and error messages show the current working directory name.

```
mkdir /tmp/i-am-the-directory
cd /tmp/i-am-the-directory/
openscad
```
Just add `assert(false);` and press `F5` which gives
```
ERROR: Assertion 'false' failed in file i-am-the-directory, line 1
TRACE: called by 'assert', in file i-am-the-directory, line 1.
```

nophead, 2019-03-17:
I think they should be i-am-the-directory/Untitled.scad as that is the default filename before a save.

RomitKumar, 2019-04-19:
A possible solution can be to save file(or prompt save file dialog box if file is being saved first time) whenever preview or render is done. The function for saving file will be called only when the file is modified. 
Should I work on this approach?

t-paul, 2019-04-19:
Forcing to save the file could be a bit surprising to the user. It should be fine to just use the default dummy file name as @nophead suggested. Although we need to make sure we are using the existing logic as that name depends on the language settings  (e.g. with `LANG=de_DE` it's "Unbenannt.scad" instead of "Untitled.scad").
I suspect the handling of that is a bit scattered in multiple places, causing that problem.

RomitKumar, 2019-04-19:
Maybe there's need to handle a special case when the program is not yet saved 

https://github.com/openscad/openscad/blob/fb335dfe57ed51a5fb8c9782a4b56bc7f26aca8a/src/expr.cc#L773-L779

locs can be set to 'Untitled.scad', in above code, if file is not yet saved once

t-paul, 2019-04-19:
Hmm, looking at that snippet, I suspect the main issue is that we have no connection to the GUI code from the expression handling. That is intentional as we do want to support compiling the engine without the GUI part. While that is not really maintained in the official source, there's just a recent project to build a non-GUI version of OpenSCAD (http://repo.hu/projects/openscad-cli/).
I guess the main question is how to handle the information transfer from main window to the core engine without breaking the separation. I seem to remember we had a similar discussion in a different context, so if we find a good solution that could be useful in other scenarios too.

jordanbrown0, 2026-03-27:
This should really get fixed, and probably isn't hard.  I'd suggest that it should simply omit the file name when the file is not saved.