openscad/openscad #2879: Improve option handling in release-common.sh

t-paul, 2019-03-17:
The options are read via different methods which makes things a bit difficult to get correct.

* `-mingw64 -v VERSION` works but produces lots of warnings from getopt about unknown options
* `mingw64 -v VERSION` does not produce warnings, but does not pickup the version number as getopt stops at the non-option mingw64 word
* `-v VERSION mingw64` works withoug warnings

thehans, 2019-03-17:
The problem is the `getopts` bash function in the script doesn't support long options, only single characters.  

I noticed this a while ago and changed the docker images for example to not set the dash before the `mingw64` and `snapshot` etc (the script just greps the args for these instead of using `getopts`), but I didn't know that the ordering of `-v` also matters: https://github.com/openscad/docker-openscad/commit/7457ee34fc96ff37cb2bbf35576cbd8c103c3c52

t-paul, 2019-03-17:
Maybe we just go with the short names then? Yes, the docker builds are where I stubled across the issue as the generated version for the RC4 builds were not correct.
Not a very urgent, I guess, but might be worth cleaning up at some point to make it less surprising.