openscad/openscad #2880: Compilation from source on Mac fails on -lqscintilla2_qt4

thboomsma, 2019-03-19:
Getting error om compilation on Mac OSX High Sierra trying to get the Post Production Carve tooling to work. Anybody had succes with it?

    $ qmake  CONFIG+="experimental carving -qt=qt5" OPENSCAD_LIBDIR=/usr/local
    $ make

`ApplicationServices are out of sync. Falling back to library file for linking.
ld: library not found for -lqscintilla2_qt4
clang: error: linker command failed with exit code 1 (use -v to see invocation)`

t-paul, 2019-03-19:
Qt4 is not supported on Mac anymore I believe (as of #2811).

t-paul, 2019-03-19:
Wait, where is that CONFIG value coming from? that looks quite unusual. What is that `carving -qt=qt5` part? If you are doing very specialized build, you really need to add a bit more context.

thboomsma, 2019-03-19:
> Wait, where is that CONFIG value coming from? that looks quite unusual. What is that `carving -qt=qt5` part? If you are doing very specialized build, you really need to add a bit more context.

Hi im really not interested in QT4 or 5 but just trying to get the Carve functionality in OpenScad going on my Mac

The ref:
https://github.com/openscad/openscad#building-for-mac-os-x
Compilation
First, run qmake openscad.pro from Qt to generate a Makefile.

On some systems, depending on which version(s) of Qt you have installed, you may need to specify which version you want to use, e.g. by running qmake4, qmake-qt4, qmake -qt=qt5, or something alike.

Then run make. Finally you might run make install as root or simply copy the 'openscad' binary (OpenSCAD.app on Mac OS X) to the bin directory of your choice.

t-paul, 2019-03-19:
OpenSCAD has currently 2 build variants, one is based on homebrew (using `macosx-build-homebrew.sh`), one is compiling everything (all dependencies) from scratch (using `macosx-build-dependencies.sh`). Which one are you using?

You have to make sure the build uses Qt5. It's obviously trying to link against a Qt4 QScintilla as the name indicates, so there's something strange going on in the build environment. That should not come from the normal OpenSCAD build as Qt4 support was dropped. Are you sure that `qmake` you call is from Qt5?

kintel, 2019-03-19:
@thboomsma `carving` isn't part of OpenSCAD; you'd have to use @peberhard's fork for that. ..and you probably have to bring that fork up to date to be able to use any updates to our build system.

Closing as not an OpenSCAD issue

closed 2019-03-19 by kintel