openscad/openscad #2882: Allow trailing comma in let() or lists

adrianVmariano, 2019-03-20:
Other languages like C++ and Python allow a trailing comma in contexts where comma separated lists occur.  This makes it easier to move code around, or add and remove things.  So for example it would be nice to be able to write

    let(
        a=<expression>,
        b=<expression>,
        c=<expression>,
    )

and then if I deleted the third expression, or exchange the order the code still works.  Similarly, it would be nice if I could write

    points =  [
               [1,1],
               [2,2],
               [3,3],
              ]

for the same reason.

kintel, 2019-03-20:
The second example is already possible, no?

I think it makes sense to extend to other list-like constructs, unless we find conflicting hidden bugs/features that people are abusing : /

adrianVmariano, 2019-03-20:
Hmm.  Yes, the second example does work.  I thought I had checked it.  The one that's really been driving me crazy is the last comma in let().  

The only other context I can think of is argument lists to functions and modules, and it would probably be nice to allow it there as well.   Are there any other list-like constructs?

adrianVmariano, 2020-12-25:
I must say that the comma situation in let() statements remains the cause of about 75% of my syntax errors.   It looks like somebody fixed this an issues a PR for it.

PR #3814 (2021-08-24): Clean up comma parsing (squashed)
This change is based on PR #2928, with the parsing of commas changed to be consistent in all cases, and updated to be current with master.

A single optional trailing comma is now allowed in all cases where commas are applicable: parameters, arguments, and vector contents.
Leading commas or extra "middle" commas are considered invalid syntax.  In order to be valid, a comma must be preceded by a valid element.  e.g. `v=[,]` and `v=[1,,3]` are not valid under this PR.

Resolves #2882
closed 2021-08-24 by t-paul