openscad/openscad #2883: Crash with for loops and difference: Assertion `op->left() && op->right()' failed.

ali1234, 2019-03-20:
Code:

```
tri = sqrt(3)/2;

module holes() {
    union() for (x = [-10:11]) for (y = [-10:11])
        translate([x+(0.5*(y%2)), y*tri, 0]) cylinder(r=0.35, h=100, center=true);
}

module other_holes() {
    union() for (x = [-30:31]) for (y = [-30:31])
        translate([0, 0, y*tri]) rotate([90, 0, 6*x + (3*(y%2))]) cylinder(r=0.35, h=100, center=true);
}

module basket() {
    difference() {
        cylinder(r=15, h=15);
        translate([0, 0, 1]) cylinder(r=14.5, h=20);
        intersection() {
            scale(3) holes();
            translate([0, 0, -1]) cylinder(r=14, h=20);
        }
        intersection() {
            scale(3) other_holes();
            translate([0, 0, -1]) cylinder(r=14, h=20);
        }
    }
}

basket();
```

Above code crashes git master (2a697e4090cf) and 2019.01-RC3.1:

```
OpenSCAD-2019.01-RC3.1-x86_64.AppImage: src/csgnode.cc:203: void CSGProducts::import(std::shared_ptr<CSGNode>, OpenSCADOperator, CSGNode::Flag): Assertion `op->left() && op->right()' failed.
```

rcolyer, 2019-03-20:
Notably this does not crash 15.03, making it a regression from there.

kintel, 2019-03-20:
My first hunch would be to check if #2343 may have introduced this.
@thehans Does this sound familiar?

rcolyer, 2019-03-20:
Here's a more minimal piece of code that reproduces the same bug.

```
difference() {
    cylinder(r=5, h=1);
    intersection() {
        for (x = [-4:3]) for (y = [3:5])
            translate([x, 0.2*y, 0]) cylinder(r=1, h=1);
        cylinder(r=1, h=1);
    }
    for (x = [-30:31]) for (y = [0:6])
        translate([0, 0, 0.2*y]) cylinder(r=1, h=1, center=true);
}
```

thehans, 2019-03-20:
Just from a quick test it doesn't crash if CSG elements limit is set high enough. (bumping default from 100,000 to 150,000 works for both examples).
So it seems that its passing the limit momentarily during normalization, but not cleaning up properly on abort. Also possibly not counting the nodes correctly during this process.

thehans, 2019-03-20:
I'm looking into fixing this.  I don't think the count was really correct in the original code either(it doesn't seem to subtract count when collapsing null terms for example).  I think ideally at the end of normalization if we have tracked nodecount correctly, then it would be equal to the results that we print to console by calling `CSGTreeNormalizer::count(node)`

@kintel Do you agree these counts should match?  The current `count` method counts `CSGOperation` nodes, but not `CSGLeaf`.  Is that what we intend to track in terms of CSG limit checking?

Once I'm confident the nodecount is correct I'll try setting the limit below that and make sure it still doesn't crash.

kintel, 2019-03-20:
The counts should match. It doesn't really matter so much exactly what we count as it's mostly for information and to avoid running out of memory or time.
The count would naturally be temporarily higher during normalization and it's the highest temporary count that we should use to limit resources.

thehans, 2019-03-21:
I was trying to fix the count, but its rather complex problem due to how `CSGOperation::createCSGNode` works, so I've left the counting the same.  I decided to rewrite the iterative faux-recursion to match the original recursive method exactly, so I don't have to debug my weird code from before.

thehans, 2019-03-21:
I normally wouldn't use goto but this is a rare case where I think it makes everything simpler and straightforward to understand.

closed 2019-04-07 by kintel